f4exb/sdrangel #493: Features request - LNB LO and Swap I/Q

makkiato83, 2020-03-25:
Hello,

the software is great! I am loving it. 

I am using an HackRF to listen to the output of a downconverter.
For this reason I am kind of missing the options "LNB LO" and "Swap I/Q", available in the input control of Gqrx.

So, unless these options are available (but I failed to find them so far) I would suggest:

**Features request:** Please add the options "LNB LO" and "Swap I/Q", available in the input control of Gqrx.

For example, my downconverter, converts the BroadCast FM band [88Mhz, 108Mhz] as follows:

f  in [88,108] -----[downconversion]----> 21.4 -(98-f)    

In other words, the band [98-10,98+10] is downshifted to [21.4 -10, 21.4 + 10] and mirrored, 
so that 11.4 Mhz corresponds to 108Mhz and 31.4Mhz to 88Mhz.

The feature "Swap I/Q" allows to basically ignore the "mirrored" complication.
The feature "LNB LO" allows to basically ignore the downshift.

With those two options I could just listen to [88Mhz,108Mhz] naturally, without having to do the math of the downconversion.

Thanks!!! Keep up the great work!

makkiato83, 2020-03-26:
My apologies: I just discovered that the feature "LNB LO" is already implemented!
It is a small "X" bottom-right of the frequency display. Not super visible, but present!

Still,  I don't know if there is a "Swap I/Q" option currently available.

f4exb, 2020-03-28:
The "LNB LO" is in fact the transverter option and indeed this is not documented in the HackRF input plugin readme (hence documentation label). This is the "X" button next to "RF Amp". it opens a new dialog with a thumbwheel and a check button. To activate transverter function you switch the check button and you can enter the frequency shift with the thumbwheel.

This is correctly documented in RTL-SDR and you can check this since it is the exact same functionnality: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/rtlsdr This is control number 5.

The I/Q swap could be an interesting addition to the transverter function because indeed for now we assume straight conversion (enhancement label).

makkiato83, 2020-03-28:
Thanks a lot for the useful answer!
Indeed I missed the "X" icon.

I take this opportunity to propose another feature:

Currently, for the HackRF, the maximal decimation factor is 64. 
The HackRF specs recommends to use 20MS/s if the CPU allows, 10MS/s otherwise, and an absolute minimum of 2MS/s.

This means that, with factor 64, it is currently not possible to "Zoom in" a lot in the spectrum waterfall.
I am, for instance, looking at FT8 signals which are 50Hz wide, and with:
-) 20Ms/s
-) 64 Decimation

I can only get about 20khz/division in the waterfall display.
So, if it is technically possible:

**Feature Request:** Add 128 and 256 Decimation factors for HackRF.

Thanks again!

f4exb, 2020-03-28:
This is theoretically possible to add decimation (and interpolation for Tx) factors beyond 64 (as powers of two). After all this is just a matter of chaining more half-band decimators (or interpolators). However this has not appeared to be necessary for a very log time and since some time (don't remember the exact version) there is a workaround with the LocalSink plugin: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/localsink in the device set where you have the HackRF and the LocalInput plugin as the device in another device set: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/localinput

This is what I am currently using to "zoom in" parts of a bigger part of spectrum of the RTL-SDR used as a IF receiver for QO-100 (after the LNB).

f4exb, 2020-06-24:
Swap I/Q control implemented in transverter dialog and REST API device settings (iqOrder) in v4.14.11

closed 2020-06-24 by f4exb