f4exb/sdrangel #494: LimeSDR: couldn't find LimeSDR device in sampling device select list.

Texas-C, 2020-03-26:
Enviroment: `ArchLinux`, LimeSuite already installed. 
Descript: Can not find LimeSDR device in select list. (other device like `rtl-sdr` is available)

More details:

When I running `SDRAngel`, I got following messages in console:

    2020-03-26 19:56:58.029 (W) PluginManager::loadPluginsDir: Cannot load library /usr/lib/sdrangel/plugins/libinputlimesdr.so: (libLimeSuite.so.19.04-1: cannot open shared object file: No such file or directory)
    2020-03-26 19:56:58.039 (W) PluginManager::loadPluginsDir: Cannot load library /usr/lib/sdrangel/plugins/liboutputlimesdr.so: (libLimeSuite.so.19.04-1: cannot open shared object file: No such file or directory)

Both of `libinputlimesdr.so` and `libinputlimesdr.so` relay on the `libLimeSuite.so.19.04-1`.

Seems it's version issue of `libLimeSuite.so`. Cause I found I already have `libLimeSuite.so `, `libLimeSuite.so.20.01-1` and `libLimeSuite.so.20.01.0` in `/usr/lib`.

Maybe link both of `libinputlimesdr.so` and `libinputlimesdr.so` to the `libLimeSuite.so`, not specific version of `libLimeSuite.so`, could solve this issue.

f4exb, 2020-03-28:
Arch Linux is not supported. In any case now Limesuite 20.01.0 is used now and it is up to you to make sure this is correctly implemented in your environment.

Texas-C, 2020-03-29:
Thx

I have solved by compile `SDRAngel`. 

But I still don't know why `libinputlimesdr.so` didn't link to the newest `libLimeSuite.so` when install by `pacman`.

And I found another problem that `SDRAngel` can not compile on both `Qt 5.14` and `Qt 5.12 LTS`. It would got a `compile error`:

    /usr/include/qt/QtCore/qglobal.h: In instantiation of ‘T qExchange(T&, U&&) [with T = 
    QDebug::Stream*; U = long int]’:
    /usr/include/qt/QtCore/qdebug.h:118:77:   required from here
    /usr/include/qt/QtCore/qglobal.h:1050:7: error: invalid conversion from ‘long int’ to 
    ‘QDebug::Stream*’ [-fpermissive]
     1050 |     t = std::forward<U>(newValue);
          |     ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
          |       |
          |       long int

It seems is Qt's bug, but I don't know which code make Qt's bug occur. ( maybe we could submit a issue report to Qt dev?)

Em.. Which version of Qt you have tried? I rember that Qt's version on  `Ubuntu 18.04 LTS` is `5.9`. I'm afraid in future version, `SDRAngel` can not be compile on such new system like `Ubuntu 20.04 LTS`.

f4exb, 2020-03-29:
Yes, the reference environment is Ubuntu 18.04 and therefore Qt version used is 5.9.5. This may become an issue when upgrading to Ubuntu 20.04 but this will be addressed in time. It seems to be a general issue in Qt so I suspect that either Ubuntu will not upgrade Qt to this version or the issue will be fixed in Qt.

Texas-C, 2020-03-31:
Thx

And I found an other interesting things.

In `wayland` enviroment, `SDRAngel` which installed by `apt` (in ubuntu) or `pacman` (in Archlinux). Can not display some graphic widget like spectrum, ATV.

However, compiled version could display thoes widgets (and works really well, Archlinux + KDE + Wayland + Qt5.9).

BTW, is it necessary to remove `sdrangel` from Archlinux's software repository? The issue that cannot find SDR is only happend on `LimeSDR` & `LimeSDR-Mini`. Other SDR like `hackrf`, `rtl-sdr` and `airspy` are still works on Archlinux (repository edition)

Texas-C, 2020-04-10:
Hi~

I found a solution to let SDLAngel could be build on `Qt 5.14`. (Tested on Archlinux & Qt 5.14.2)
Edit `/usr/include/qt/QtCore/qdebug.h` line `118`:

    QDebug( QDebug &&other ) noexcept : stream{qExchange(other.stream, nullptr)}{}

to:

    QDebug( QDebug &&other ) noexcept : stream{qExchange(other.stream, (QDebug::Stream*)(nullptr))}{}

I guess this issue is cause compiler was mistaken seem `nullptr` as `long int` type, not `QDebug::Stream*` pointer type. So I forced to convert `nullptr` as `QDebug::Stream*` type. Then compile successed.

Texas-C, 2020-04-10:
As the [post](https://bugreports.qt.io/browse/QTBUG-83435?focusedCommentId=506868&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-506868) said.

This error may cause by `cm256/gf256.h`:

    #ifndef nullptr
        #define nullptr NULL
    #endif

f4exb, 2020-04-11:
Indeed this is very wrong. This issue has been opened in cm256: https://github.com/f4exb/cm256cc/issues/13

closed 2020-05-16 by f4exb