f4exb/sdrangel #495: Timming and dropped audio frame issue with HackRF SDRangel Windows.

cjflnyc, 2020-03-28:
Good day, I use HackRF Windows 7 64 bits and SDRAngel to transmit with HackRF. The Wideband FM modulator is what I use specifically, I been trying to experiment with digital modes, modems etc, And I seem to be having what i'm guessing some sort of Timing or dropped frames issue on this mode with HackRF.

The receiver station can not get a lock on any digital signal modulated with SDRangel( I use virtual audio cables to route the audio from one program to the other)

The reason I know this is most likely a software issue is that when I replace the SDRAngel/HackRF setup with an analog transmitter with output from soundcard connected.

HackRF and Sdrangel as a receiver, picks it up fine and locks. and I can play with digital modes.

I read that for some digital modes. A timing issue of 5us and even some dropped frames in the audio to such a small extent that we can't notice it by ear, Can effect this as well. (cause loss of lock on some modes)

So maybe in a future version some options to address this perhaps, More control of the audio/HackRF unit.?

f4exb, 2020-11-21:
Possibly related to https://github.com/f4exb/sdrangel/issues/429

f4exb, 2020-11-22:
Audio input rework implemented in v6.1.0

closed 2020-11-22 by f4exb