f4exb/sdrangel #497: GLScope: marker feature

f4exb, 2020-03-29:
This is a request received directly and is of interest. To enhance GLScope functionnality it would be nice to have a basic marker functionality within the scope window so that precise values can be displayed for a given point (clicked) in the graph. A picture is worth many words so here is one to get the idea:
[image]

Of course there should be a mean to erase the marker (right click?).

f5sfu, 2020-03-29:
Hi F4EXB,
that looks very very nice. Will the pointer be free or attached to the curve ? as we can see in some simulation plots ?
In both cases, that's a great feature :)
Is it easy to provide automatic max min reporting ? As an example, user would directly get its deviation evaluation ?
Thanks a lot for this, I guess it can have much more applications than just max FM deviation.
Right click to remove pointer  seems fine :)
Best 73

f4exb, 2020-03-29:
Hi F5SFU :smile: 
I think I will make it simple however after a scond thought there could be two points so that the difference x1 - x0 and y1 - y0 can be displayed. To be arranged in a stacked fashion like:
  - first left click: mark first point
  - second left click: mark second point
  - third left  click: change second point
  - right click: remove second point
  - ...

f5sfu, 2020-03-29:
das wäre super !

f4exb, 2020-07-06:
Implemented in v5.7.8 and v4.14.15

closed 2020-07-06 by f4exb