f4exb/sdrangel #498: Give me the document that helps repair the. Recorder player.

gaspar1987, 2020-04-04:
I'd like to know why the recorder isn't working.
Because I tested it.
And I don't see the message writing it.
Written reader save and I don't see in the document the file corresponding to the recording. He hasn't recorded since version 4 of.
Rangel.
Disturbing, she.
I ask why sir. It doesn't work giving me an answer that can satisfy me with.
Fix it.
Because.
It's a bit of a shame. It works on all software. But the recorder is still.
It's broken down.

gaspar1987, 2020-04-04:
[image]

gaspar1987, 2020-04-04:
I would like you to give me possible software to repair it or the specific data that can decrypt.
Where to fix this famous one?
Recorder, because since.
Version three, I got in trouble after that.

f4exb, 2020-05-14:
The screenshot shows you have no .sdriq file in the folder where you are trying to find it period.

closed 2020-05-14 by f4exb