f4exb/sdrangel #501: PlutoSDR in Version 4.14.3?

MichaelRenner, 2020-04-05:
Moin,

I used to use a networked PlutoSDR with SDRangel version 4.11.10. Worked fine. Now I updated to version 4.14.3, but the PlutoSDR is not longer recognized. After downgrading to 4.11.10 (luckily I still had the deb) every is fine again.

From the start messages:
```
2020-04-05 09:50:40.442 (D) PluginManager::registerSampleSource  PlutoSDR Input  with source name  sdrangel.samplesource.plutosdr  and hardware id  "PlutoSDR"
2020-04-05 09:50:40.442 (D) PluginManager::registerSampleSink  PlutoSDR Output  with sink name  sdrangel.samplesink.plutosdr  and hardware id  "PlutoSDR"
```

So, what have I do to use my PlutoSDR with the current version?

Thanks

f4exb, 2020-04-05:
No reason it should not be there. Here are the relevant extracts of the build log in Appveyor (57-dev):

```
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/libiio.so
...
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/libplutosdrdevice.so
[00:32:39] -- Set runtime path of "/home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/libplutosdrdevice.so" to "/usr/lib/sdrangel"
...
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/plugins/libinputplutosdr.so
[00:32:39] -- Set runtime path of "/home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/plugins/libinputplutosdr.so" to "/usr/lib/sdrangel"
...
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/plugins/liboutputplutosdr.so
[00:32:39] -- Set runtime path of "/home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/plugins/liboutputplutosdr.so" to "/usr/lib/sdrangel"
...
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/pluginssrv/libinputplutosdrsrv.so
[00:32:39] -- Set runtime path of "/home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/pluginssrv/libinputplutosdrsrv.so" to "/usr/lib/sdrangel"
...
[00:32:39] -- Installing: /home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/pluginssrv/liboutputplutosdrsrv.so
[00:32:39] -- Set runtime path of "/home/appveyor/projects/sdrangel/debian/sdrangel/usr/lib/sdrangel/pluginssrv/liboutputplutosdrsrv.so" to "/usr/lib/sdrangel"
...
```
So problem is with your installation.

MichaelRenner, 2020-04-05:
Thank you for your answer.
I checked it twice. With version 4.11.10 the PlutoSDR is found, with 4.14.3 not. In between there were only "dpkg -i ...". Where can I search? I made a new user - there is no PlutoSDR for that either. Where else could I search, what else could I try?

vy 73 de Micha, DD0UL

MichaelRenner, 2020-04-05:
I did a little research.

A locally connected PlutoSDR is recognized, also with 4.14.3, a PlutoSDR in the network is not recognized by 4.14.3, but from the previous installation 4.11. I installed sdrangel on windows 10 (for the first time). Here we have same effect: It works locally connected to USB, but not in the network.

vy 73 de Micha, DD0UL

f4exb, 2020-04-06:
Yes the behavior has changed since v4.12.3 actually. This is explained here:https://github.com/f4exb/sdrangel/releases/tag/v4.12.3

To summarize you have to give the address of the Pluto in the device parameters. This is to avoid unbearably lengthy search of the network for a Pluto that does not exist in most of the user cases.

MichaelRenner, 2020-04-06:
Moin,

thanks for the reply and the hint for the change. I liked the autodiscovery, but it is as you write: only very few users will be able to use it.

73 de Micha, DD0UL

closed 2020-04-06 by MichaelRenner