f4exb/sdrangel #503: error in cmake

fe8769a, 2020-04-14:
Hello from France

error in cmake  not found after many requests on internet
Using :+1: 

cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..

here is error message 

-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
CMake Error at /usr/share/OpenCV/OpenCVConfig.cmake:183 (message):
  opencv_imgcodecs is required but was not found
Call Stack (most recent call first):
  CMakeLists.txt:326 (find_package)

-- Configuring incomplete, errors occurred!
See also "/opt/build/sdrangel/build/CMakeFiles/CMakeOutput.log".

thanks for help

loewal, 2020-04-14:
Perhaps not the same: cmake as discribed in dev-section and of course all the libs.
Make gives this:
plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/build.make:121: recipe for target 'plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/freedvdemodsink.cpp.o' failed
make[2]: *** [plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/freedvdemodsink.cpp.o] Error 1
CMakeFiles/Makefile2:4876: recipe for target 'plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/all' failed
make[1]: *** [plugins/channelrx/demodfreedv/CMakeFiles/demodfreedv.dir/all] Error 2
Makefile:171: recipe for target 'all' failed
make: *** [all] Error 2

Is there someone who is willing to look at these errors?
Thnx!

loewal, 2020-04-14:
By the way: compiling under Lubuntu (DragonOS LTS)

f4exb, 2020-04-18:
Title changed to reflect that actually OpenCV should not be required as modatv is the only plugin using it and is already optional.

f4exb, 2020-04-20:
Implemented in v4.14.4 and v5.4.0

closed 2020-04-20 by f4exb