f4exb/sdrangel #506: LimeSDR Mini Antenna Select is mislabelled

justindthomas, 2020-04-18:
The options for selecting an antenna using the LimeSDR input plugin are: No, Hi, NC,  Lo, T1, T2.

The hover description is:

"Antenna Select: No: None, NC: Not Connected, Lo: 700:900M, Hi: 2:2.6G, Wi: wideband, T1: Tx1 LB, T2: Tx2 LB"

The labels don't match up internally (i.e., there is no "Wi" in the dropdown). And none of it seems to make sense for the LimeSDR MIni - there are only two connectors, a Tx and an Rx.

I've just been selecting "Lo" and it seems to work okay.

I started to dig through the code to see if I could submit a PR myself, but I haven't done C++ in 15 years or so and figured opening an issue might be quicker.

f4exb, 2020-04-19:
AFAIK the Mini does have internal switches that select different matching circuits behind the physical Rx and Tx ports.

f4exb, 2020-04-19:
Commit https://github.com/f4exb/sdrangel/commit/50d39f8 refers

justindthomas, 2020-04-19:
That's great, thanks for looking in to it. Is there a way to incorporate the "TX" and "RX" labeling in the options? The ports are physically labeled as such and that would be helpful especially for new folks. Like "RX High, RX Wide, TX High, TX Wide."

Or would that make it too much of a snowflake from the larger LimeUSB?

f4exb, 2020-04-19:
The "Rx" or "Tx" labeling does not bring anything since you already know by the plugin on which side you are.

justindthomas, 2020-04-19:
Got it. I just installed the latest dev branch. That looks great. Thank you!

f4exb, 2020-04-20:
Now implemented in v4.14.4 and 5.4.0

closed 2020-04-20 by f4exb