f4exb/sdrangel #508: remote tcp connections top API

XReyRobert, 2020-04-20:
I've quickly looked around but didn't see how to enable remote connections to API of the running GUI client ? It's only listening on localhost ip, I'd like it to listen on others ips...

f4exb, 2020-04-20:
https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/remoteinput/readme.md
https://github.com/f4exb/sdrangel/blob/master/plugins/samplesink/remoteoutput/readme.md
https://github.com/f4exb/sdrangelcli
https://github.com/f4exb/sdrangel-docker/tree/master/sdrangel

f4exb, 2020-04-22:
Or maybe if your network interface address is say 192.168.0.5 launch SDRangel with -a option like this: `sdrangel -a 192.168.0.5`. (`sdrangel -h` displays all available options)

What you are trying to achieve is very unclear.

f4exb, 2020-05-20:
Question answered.

closed 2020-05-20 by f4exb