f4exb/sdrangel #509: Incorrect behavior where enabled transverter settings

exception13, 2020-04-22:
**Steps for reproduce:**

1. Start application
2. Set transverter frequency (-120MHz in my case) and enable frequency translation
3. Set LO frequency (7MHz)
4. Restart application

**Expected Result:**
After restart application LO frequency must be 7MHz

**Actual result:**
After restart application LO frequency set to minimum receiver frequency (10MHz for LimeSDR Mini)

The same behavior is observed when switching presets with transverter frequency translation settings. If you select a preset with transverter frequency translation enabled, the frequency is set to the minimum until you just open and close the transverter frequency translation settings window.

f4exb, 2020-04-25:
Reproduced. In fact this is due to the frequency limits not being applied before the frequency display is applied in the displaySettings() method of the GUI. Note that this issue may affect other plugins.

f4exb, 2020-04-25:
PlutoSDR: OK

f4exb, 2020-04-28:
Implemented in v4.14.5 and v5.5.0

closed 2020-04-28 by f4exb