f4exb/sdrangel #510: Unreachable Network-PlutoSDR crashes SDRangel

MichaelRenner, 2020-04-26:
Moin,

I configured my network based PlutoSDR in "Preferences | Devices | User arguments". This works fine so far.
But: The PlutoSDR is only one of my SDRs and it's not running all the time. But If the PlutoSDR is not reachable sdrangel crashes:
Segmentation fault (core dumped)
Due the fact that sdrangel won't start it's not possible to reconfigure the device. I found the configuration file .config/f4exb/SDRangel.conf. Unfortunately it is not human readable, I only can delete it, but it's not possible to edit it.

Would it be possible to handle this "device not found" exception? Maybe by disablering such a device or so?
```
2020-04-26 11:35:22.147 (D) DeviceEnumerator::addNonDiscoverableDevices: start
2020-04-26 11:35:22.147 (D) DeviceEnumerator::addNonDiscoverableDevices: Rx: PlutoSDR[0:0] user defined
2020-04-26 11:35:22.147 (D) DeviceEnumerator::addNonDiscoverableDevices: Tx: PlutoSDR[0:0] user defined
2020-04-26 11:35:22.148 (D) MainWindow::MainWindow: select SampleSource from settings or default (file input)...
```

renner@solaris:~$ sdrangel --version
2020-04-26 11:34:14.043 (W) Attribute Qt::AA_EnableHighDpiScaling must be set before QCoreApplication is created.
SDRangel 4.14.3

Thanks

f4exb, 2020-05-20:
The root cause seems to be that in the `DeviceEnumerator::addNonDiscoverableDevices` it takes for granted that a "non (automatically) discoverable" device is always reachable in the system. As this is not always the case it should probe it in some way before adding it to the list of available devices thus avoiding adverse effects when trying to use the device afterwards.

f4exb, 2020-05-20:
As a workaround you can edit `~/.config/f4exb/SDRangel.conf` which is still text even if non readable and remove the line starting with `current=` completely. You will loose your current defaults but be able to start again.

f4exb, 2020-05-20:
In fact it is just the opening of the device and derived pointer(s) usage that is not handled properly when open fails in `PlutoSDRInput`. Same is expected in `PlutoSDROutput` and probably other device handlers. However this sort of user parameters are used for Pluto only so other use cases may never occur.

f4exb, 2020-05-20:
Implemented in v5.6.1 and v4.14.7

closed 2020-05-20 by f4exb