f4exb/sdrangel #512: Libaccodec57 not available in 20.04

alphafox02, 2020-04-30:
Trying to install deb package on 20.04, but libavcodec58 is present, not 57, and so it will not install. It sounded like others may have sdrangel running on 20.04.

f4exb, 2020-05-01:
Have you read the doc? The provided .deb package is for 18.04. You may have some success at compiling in 20.04 although I only tried the server variant on arm64 (Rpi4).

alphafox02, 2020-05-01:
Figured that was the case. I was able to follow your page on building it from source, however I stopped short of building all the parts for each SDR as I had installed them from PPA and Apt. I went ahead and just compiled the sdrangel part with no issue, of course I had no access to any SDR though. Would you recommend uninstalling all the various rtlsdr, hackrf, soapy etc drivers and then following your page on building them all from source?

f4exb, 2020-05-16:
Yes , you have to build dependencies from sources also to guarantee the proper versions are used.

alphafox02, 2020-05-16:
Awesome. I’ll try it out! Any chance you tested on Debian Bullseye (testing)?

f4exb, 2020-06-20:
From v4.14.10 on the default .deb package in the releases section is based on 20.04 so this should not be an issue anymore.

closed 2020-06-20 by f4exb