f4exb/sdrangel #517: "RFamp" Button reduces TX power

hansen5604, 2020-05-03:
When I run my new HackRF One to a meter with SDR Angle in NBFM mode and I turn the VGA power up all the way without the RF button clicked I get only -10dbm. when you click on RF the output goes down. It should go up. Normal output is around 10dbm at 1296MHZ the manual says.

I saw a you tube video of a guy talking to another ham with SDR Angle and he used the RF button to get more power out.
Please advise.
Nice Job on this software..
Skip Hansen N0CTR
skipn0ctr@gmail.com

f4exb, 2020-05-03:
The final amp stage on the HackRF is rather fragile. Can you check it is actually working by other means? Not sure the HackRF library provides this or you may use GNU radio. On my part this is unreproducible.

hansen5604, 2020-05-03:
I have verified with a watt meter this is happening. I received the power
meter yesterday to validate my findings and I was correct
I received my HackRF one  a month ago. I did not upgrade the firmware and
am new to this device. It seems complicated to update and i don't want to
brick it.
GNU Radio on Lenox running GQEX does not allow transmit neither does SDR
Console for the Hack RF one.
I am building a 1296 EME station and have got very far. So far.

Would firmware cause this?

f4exb, 2020-05-04:
From the SDRangel point of view all I can say is that the v2018.01.1 version of libhackrf is used to build the Ubuntu binary and also with sdrangel-docker and in the compile instructions. Therefore the firmware should match this version of the library. HackRF is pretty stable now so this is probably the case but may be worth to verify.

Also I noticed that the "RF amp" setting may not be initialized properly when loading a preset so if this is the case you may need to toggle it off and back on.

hansen5604, 2020-05-04:
Thank you.
 I have checked my new power meter it was off calibration. So now with a
amplifier at 8.5db on the output of the hackRF one in SSB mode using CW
with my call at 1296Mhz I get 200mW approximately at 1296Mhz. stable.
The RF button reduces the power on my setup when checked by a factor of
10db.
So far I can work with this. I will let you know more as I go as I have a
SWR Bridge and, more amplifiers and a directional coupler with attenuators
coming.Plus a 18W Mitsubishi chip RA181213G 18Wamp and dish to build.

hansen5604, 2020-05-07:
Good morning. I will have a directional coupler soon to validate power and
SDR Angle RF button from my side.

Question:
How can one use the SDR Angle Xmit and receive like a transceiver,
  key the xmit, receive turns off. release xmit, receive is on?

hansen5604, 2020-06-16:
Please see my attached picture:  Hack RF SDR1296 station EME 2 x 30W amps
and 2 x 31 element circular and yagi W dual LNA's

I need to control an amplifier TX circuit.
Also do you have a vox planned for SSB?
How can I run JT-65 mode with SDRAngle?
Thank you for your software and help.
Skip Hansen
N0CTR Brainerd, MN USA

f4exb, 2020-06-20:
- All controls are to be made by an external program that can be piloted and can pilot SDRangel using the REST API.
- Vox for SSB makes no sense: when there is no modulation there is no RF either
- You can pipe audio in and out using `pulseaudio`. However the Tx side (audio in) seems to have buffering issues.

hansen5604, 2020-06-22:
Thank you. I will look into Rest.I have confirmed my original hack rf one is defective. I purchased another unit and it works perfectly with the RF button.Question how would I use JT65 or MT65 with Sdr Angel?

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/f4exb/sdrangel/issues/517#issuecomment-646961842",
"url": "https://github.com/f4exb/sdrangel/issues/517#issuecomment-646961842",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

f4exb, 2020-06-23:
User confirmed the HackRF unit was defective hence issue can be closed

closed 2020-06-23 by f4exb

hansen5604, 2020-07-30:
How can I run JT65 with 2 hack RF Ones on a windows 10 machine SDR Angel
running two instances to control transmit and receive?
N0CTR

hansen5604, 2020-09-09:
See pics please. 1296 JT65 Station Complete w dual SDR Angel. computers, 1
RPI4, 1 Win 10. 1 win 8 for WSJT 10.0 JT65C.
Dual Hack RF Ones with modified TX RX hardware connections. 2 x 30 watt
amps. 2 x 31 element yagis. Soon to have dual 55 elements..
Fully automatic!!

Thank you again for the great software..
Skip