f4exb/sdrangel #518: Feature request: VOX (or PTT via COM port)

Deimos1994, 2020-05-06:
Hello, I think it would be awesome to have a voice activation feature when using NFM modulator.
For example, together with this soft http://f6dqm.free.fr/soft/simplex/en/simplex.htm it would allow to make a simple echo repeater. An alternative solution would be to make virtual COM port so other programs could control SDRAngel, by control I mean simple PTT via COM port with RTS signal (tx command).
Also, I have another request but this time I want SDRAngel to control other devices via COM port. I want this in order to control external amplifiers (RX/TX). Basically I want what WSJT-X has... an ability to choose PTT method via COM port.
All these features seem very useful from my perspective, but maybe it is just me.
P.S. Thank you for awesome software, it is very useful, I made a lot of QSO with WSJT-X+SDRAngel+HackRF+AMP, I receive with RTL-SDR. 
UR4ULQ, 73!

f4exb, 2020-05-09:
There is no need for stone age technology like cotnrol via COM ports. At least not directly. With SDRangel control is made via REST API. SDRangel can also synchronize other programs (and therefore devices behind them) via the same REST API syntax. 

Please take time to read the documentation. You can access interactive REST API documentation with opening a browser at http://localhost:8091 (if program is started with all defaults).

PTT has to be controlled via REST API also. For VOX you would have to implement it externally also.Examples of PTT control are here:
  - https://github.com/f4exb/sdrangel/blob/master/scriptsapi/ptt_active.py
  - https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/ptt.py 

To conclude this is not going to be implemented in any way because:
  - control already exist via REST API and is infinitely richer and modern than COM ports
  - only DSP related are to be implemented within the software all ancillary controls can be made via external programs interfacing with REST API.

Deimos1994, 2020-05-09:
Thank you very much for your response. I did not know that SDRangel has so rich REST API, that is great. Maybe there is no need for stone-age technology like control via COM, but it the easiest solution I found for a DIY power amp control (USB-to-serial adapters are cheap). I'll try to use the examples to write some python code for my needs, thank you.

closed 2020-05-09 by Deimos1994