f4exb/sdrangel #519: SoapySDR not found during CMake config

ashley-b, 2020-05-09:
SoapySDR is not found during CMake config, if I set SOAPYSDR_DIR to force its location it finds it.
Its strange as I do have a .pc file installed at /usr/local/lib64/pkgconfig/SoapySDR.pc, but it is not found.
The fall back find does not find it also. I think there is a fault with the logic
https://github.com/f4exb/sdrangel/blob/93c13e449c8b65f3139b26a1bb505513c5773d9e/cmake/Modules/FindSoapySDR.cmake#L14
Should be
```
 if(NOT LIBSOAPYSDR_PKG_FOUND OR (DEFINED SOAPYSDR_DIR))
```
Which fixes the issue for me.

ashley-b, 2020-05-09:
Just a update, if I remove the lower case "soapysdr" version, it is found.
https://github.com/f4exb/sdrangel/blob/93c13e449c8b65f3139b26a1bb505513c5773d9e/cmake/Modules/FindSoapySDR.cmake#L12
LIBSOAPYSDR_PKG_FOUND seams to only get set when both are found far as I can tell.

f4exb, 2020-05-10:
This seems to make sense. However as this is a sensitive area it must be checked that he suggested fix works in all circumstances. The same configuration exists for `FindBladeRF` and `FindLibIIO`.

For the second issue may be `pkg_check_modules` should be replaced by `pkg_search_modules` since the [doc](https://cmake.org/cmake/help/v3.9/module/FindPkgConfig.html) says `pkg_search_module` is:

> Same as pkg_check_modules(), but instead it checks for given modules and uses the first working one.

Conversely `pkg_check_modules` needs to find all given modules.

f4exb, 2020-06-20:
Implemented in v4.14.10 and v5.7.1

closed 2020-06-20 by f4exb

ashley-b, 2020-07-04:
Thanks for the fix.
Just to let you know that the matching endif is mismatched from your commit https://github.com/f4exb/sdrangel/commit/f86cd887f34a85716ad12a62b9e265331ceeef80
CMake warning
```
CMake Warning (dev) in cmake/Modules/FindSoapySDR.cmake:
  A logical block opening on the line

    sdrangel/cmake/Modules/FindSoapySDR.cmake:14 (if)

  closes on the line

    sdrangel/cmake/Modules/FindSoapySDR.cmake:50 (endif)

  with mis-matching arguments.
```

Rua, 2020-07-11:
When I fix the `endif`, it still does not detect SoapySDR for me. The only output I get is
```
-- Checking for one of the modules 'soapysdr>=0.4.0;SoapySDR>=0.4.0'
```

stevefalco, 2020-11-04:
I too had trouble getting soapy to be recognized.  I tried the following, which worked for me:

cmake -D SOAPYSDR_DIR=/usr/local/ ..