f4exb/sdrangel #520: sdrangel wont start in win 10

khan2018, 2020-05-12:
Hi 
I downloaded  the latest version 4.14.5 exe file. When I start it says loading and then nothing appears.

Thanks,

f4exb, 2020-05-14:
The splash messages at startup are the following:
```
    splash->showStatusMessage("starting...", Qt::white);
    splash->showStatusMessage("starting...", Qt::white);
    splash->showStatusMessage("allocate FFTs...", Qt::white);
    splash->showStatusMessage("allocate FFTs...", Qt::white);
    splash->showStatusMessage("load settings...", Qt::white);
    splash->showStatusMessage("load plugins...", Qt::white);
    splash->showStatusMessage("load file input...", Qt::white);
    splash->showStatusMessage("load current preset settings...", Qt::white);
    splash->showStatusMessage("update preset controls...", Qt::white);
    splash->showStatusMessage("finishing...", Qt::white);
```
None of them says "loading" so your issue is badly formulated and not factual. Moreover there are no reports that would indicate a general issue so it must be due to a particular configuration in your case. If you are not more specific this issue will be closed.

khan2018, 2020-05-15:
Yes I meant the same splash messages.
This error occurs when I have multiple Pluto devices connected to the same host. I can see two different devices with two separate addresses but sdrangel fails to start after the splash messages. Nothing happens.

f4exb, 2020-05-15:
There is no auto detection of Pluto network devices anymore. Moreover I don't know if this has ever worked in Windows since it is based on Linux avahi. You have to specify the Pluto address using the Praferences > Devices > User arguments menu from the main window.

f4exb, 2020-06-20:
Question answered

closed 2020-06-20 by f4exb