f4exb/sdrangel #521: Upgrade to Ubuntu 20.04

f4exb, 2020-05-14:
Now 20.04 image is available in Appveyor. It is also present in Docker hub for amd64 and arm64. So all lights are green for an upgrade.

f4exb, 2020-05-14:
Not possible yet because of XTRX incompatibility:
```
#60 105.3   Could NOT find PythonInterp (missing: PYTHON_EXECUTABLE) (Required is at
#60 105.3   least version "2.7")
```
https://github.com/xtrx-sdr/liblms7002m/blob/91b5107a11eb5335bbaf02ac0c744ea58d2a0b74/CMakeLists.txt#L44

f4exb, 2020-05-15:
Other blocking point: sudo always asks for passwords despite the correct configuration in `/etc/sudoers`

f4exb, 2020-05-16:
Updated XTRX version in my Github: https://github.com/f4exb/images
Using this version of XTRX I was able to compile and run in a 20.04 VM

f4exb, 2020-05-20:
My changes in XTRX have been merged so this blocking point is lifted. Opened issue in Appveyor that has been classified as "bug" so expecting fix in next image release: https://github.com/appveyor/ci/issues/3434

f4exb, 2020-06-02:
Problem with Appveyor has been fixed so it could be possible to release a 20.04 package at the next release.

BX2ABT, 2020-06-08:
Problem when compiling:

```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Multimedia"
  with any of the following names:

    Qt5MultimediaConfig.cmake
    qt5multimedia-config.cmake

```
Problem is that there is no libqt5multimedia-dev in either Ubuntu 20.04 or 19.10.

f4exb, 2020-06-19:
I guess you are trying to install 18.04 .deb on a 20.04 system. This will not work.

f4exb, 2020-06-19:
Problem out of scope. `Qt5Multimedia` is not referenced in any of the cmake files.

f4exb, 2020-06-20:
Since v4.14.10 the .deb packages in the release section are exclusively based on 20.04. Converting the dev environment to make sure everything is fine with 20.04 is the last step

f4exb, 2020-08-12:
After trying Lubuntu 20.04 for some time I must say I am quite dissatisfied and disappointed with it. There are many bugs and too much details I don't like. Now considering Debian Buster + LXDE.

f4exb, 2020-08-19:
After some trial period and despite a remaining issue with AirspyHF+ the new reference environment is now Ubuntu 20.04. This has been specified in the Wiki. This issue can be closed.

closed 2020-08-19 by f4exb