f4exb/sdrangel #523: question: Android client for sdrangelsrv ?

satfan52, 2020-05-16:
I finally installed a dish on my roof to point to Oscar100 .To find the very weak Oscar100 signal west of the very powerful Astra 28.2 signal,  I need to use an sdr receiver tuned on the Oscar100 transponder frequency. I wish to avoid bringing a PC on my roof for that purpose, so it would be convenient if there was an Android client that could connect to the server version of sdrangel (ie sdrangelsrv) so I can visualize the spectrum of on my phone. Please let me know if such a suitable Android client exists, I checked the Android App store and I didn't see suitable software (at a first glance). Thanks

f4exb, 2020-05-16:
No there is no Android client nor plans to have one. Supposedly Qt5 code can be ported to Android but I tried that a long time ago and gave up. Instead you can use SDRangel GUI variant remotely to process the I/Q flow coming from a remote instance of SDRangelsrv. 

Use the RemoteSink plugin on SDRangelsrv and RemoteInput on SDRangel GUI and you are done. Controlling the instance of SDRangelsrv can be done using the SDRangelCli web application: https://github.com/f4exb/sdrangelcli.

THe very recent version v5.6.0 on the v5 branch adds the possiblity to run a "spectrum" server based on websockets to which a client can be connected to process or display PSD samples. A web spectrum visualization client is available here: https://github.com/f4exb/sdrangelspectrum

satfan52, 2020-05-17:
Many thanks Edouart. Ok so in fact, it looks like I will have to bring  a small PC on the roof.  Now that it is clarified,  I actually have 4 options :
option 1: bring my plutosdr on the roof with the PC running sdrangel 
option 2:  leave my plutosdr in the shack and connect to it over IP  from sdrangel on the roof
option 3: leave my plutosdr in the shack, run an instance of sdrangelsrv in the shack and to connect to it from  an instancee of sdrangelcli running on my roof PC
option 4: same as option 3 except that I run an instance of sdrangelspectrum on the roof instead of sdrangelcli. This requires me to upgrade to v 5.6 first though. The advantage is that I don't need to run a "headless sdrangelsrv" instance in the shack, I can just run the full V5.6 sdrangel instance with GUI.

Correct ? 
All other things being equal, Is there is a difference of performance between option 2, 3 and 4 ? Maybe some options require less network bandwidth, have a lower latency (delay) or are more reliable to use over a wifi network . After all this all about dish pointing so latency is important (I think ?)

closed 2020-05-19 by satfan52