f4exb/sdrangel #524: Dependency on libgcc-s1 but no such package exists in Bionic repo

kylex2, 2020-05-20:
Hello,

I downloaded release v4.14.6 and tried to install the deb package but without success. One of the dependencies doesn't exist in [Ubuntu Bionic repositories](https://packages.ubuntu.com/search?suite=bionic&keywords=libgcc-s1).

>  sdrangel depends on libgcc-s1 (>= 4.0); however:
>   Package libgcc-s1 is not installed.

Equivalent package for Bionic seems to be libgcc1 while for Focal and newer it is called libgcc-s1. The file sdrangel_4.14.6-1_amd64.buildinfo also contains other packages not available for 18.04. Is the build host really on vanilla Ubuntu 18.04?

Thanks

alphafox02, 2020-05-22:
Same issue with libgcc1 and 18.04 here. I’ve stayed on an older release for now.

vk6flab, 2020-05-24:
To make sure I wasn't going crazy I used a docker image, ubuntu:18.04 as the basis for installation. This breaks with 

> sdrangel : Depends: libgcc-s1 (>= 4.0) but it is not installable

The ubuntu package search tool also confirms that the libgcc-s1 package doesn't exist in bionic, bionic-updates or bionic-backports.

I attempted to install using a ubuntu:20.04 container and that created a whole different list of issues:

> The following packages have unmet dependencies:
>  sdrangel : Depends: libavcodec57 (>= 7:3.4.6) but it is not installable or
>                      libavcodec-extra57 (>= 7:3.4.6) but it is not installable
>             Depends: libavformat57 (>= 7:3.4.6) but it is not installable
>             Depends: libavutil55 (>= 7:3.4.6) but it is not installable
>             Depends: libopencv-core3.2 but it is not installable
>             Depends: libopencv-imgcodecs3.2 but it is not installable
>             Depends: libopencv-imgproc3.2 but it is not installable
>             Depends: libopencv-videoio3.2 but it is not installable
>             Depends: libswresample2 (>= 7:3.4.6) but it is not installable
>             Depends: libswscale4 (>= 7:3.4.6) but it is not installable
>

alphafox02, 2020-05-24:
Same here, think I mentioned it in another post about 20.04. I thought maybe it was just me, but seeing as you are having the same exact problem on both 18.04 and 20.04 there’s got to be something else up. I think I’ve had to stay on 57 (that number comes to mind on the zip file) for 18.04 and haven’t installed it yet on 20.04.

janoleholm, 2020-05-29:
Same here too, have tried the oldest binary on the master branch: sdrangel_4.14.4-1_amd64.deb, but same issue. In my notes I can see that 4.14.0-1 was able to install. But that version is only available in source.

rikeanimer, 2020-06-04:
Just got my Pluto and had same issue.

I solved it this way in 18.04:

[tested it with both releases 4.14.7 and 4.14.8]

* first I removed ~/.config/f4exb directory and my initially installed version of sdrangel from snap
  * snap remove sdrangel
  * rm ~/.config/f4exb -r ]
* download gcc-10-base
  * wget http://mirrors.edge.kernel.org/ubuntu/pool/main/g/gcc-10/gcc-10-base_10-20200411-0ubuntu1_amd64.deb
* download libgcc-s1
  * wget http://mirrors.xmission.com/ubuntu/pool/main/g/gcc-10/libgcc-s1_10-20200411-0ubuntu1_amd64.deb
* download a release deb of sdrangel, I tried 4.14.7 and 4.14.8 and both worked

* install gcc-10-base first, the libgcc-s1, then sdrangel release deb
  * sudo dpkg -i ~/Downloads/gcc-10-base_10-20200411-0ubuntu1_amd64.deb
  * sudo dpkg -i ~/Downloads/libgcc-s1_10-20200411-0ubuntu1_amd64.deb
  * sudo dpkg -i ~/Downloads/sdrangel_4.14.8-1_amd64.deb 

It works! First time I got the Pluto happy with ubuntu and sdrangel.

For reference
  * uname -r
5.3.0-53-generic
  * lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

f4exb, 2020-06-20:
I am afraid there is no proper solution to this other than upgrading to 20.04 (see #521). It vaguely seems it is related to the build in Appveyor that installs `gcc-9` and `gcc-10` in the build environment that are not normally available in 18.04. This introduces a "hidden" dependency to this library.

f4exb, 2020-06-20:
Set as duplicate of #521

f4exb, 2020-06-20:
Else you have to use this workaround before attempting to install sdrangel package:
```sh
sudo apt-get install software-properties-common
sudo apt-get update
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt install gcc-10
```
Ref: https://askubuntu.com/questions/1192955/how-to-install-g-10-on-ubuntu-18-04

But I will remove 18.04 package from releases as this is completely non-standard.

vk6flab, 2020-06-20:
> But I will remove 18.04 package from future releases as this is completely

Not sure if what you're describing is correct. My comments about using
Docker showed the output for both a vanilla 18.04 and 20.04 installation.

Neither is installable.

I think that there's an issue with the build system that is loading a
non-standard library that becomes an un-installable dependency.

I'm pretty sure that installing a binary package should not require that
gcc-10 is installed. (Unless the packaging script starts doing funky
compilation steps, but even that seems far-fetched.)
--
finger painting on glass is an inexact art - apologies for any errors in
this scra^Hibble

()/)/)() ..ASCII for Onno..

f4exb, 2020-06-20:
It works in 20.04 because `libgcc-s1` is in the standard distribution.

vk6flab, 2020-06-20:
I've just attempted to build the following Dockerfile:

```
FROM ubuntu:20.04
# This dockerfile is not intended for use in production
RUN apt update
RUN apt -y upgrade
RUN apt install busybox
RUN busybox --install
RUN mkdir /appdir
WORKDIR /appdir

# Get latest release
RUN wget https://github.com/f4exb/sdrangel/releases/download/v4.14.7/sdrangel-107-master.tar.gz
RUN tar -zxf sdrangel-107-master.tar.gz

# Attempt to install it
RUN apt -y install ./sdrangel-107-master/sdrangel_4.14.7-1_amd64.deb
```

It gives the following error at step 10/10, the `apt -y install` step:

```
Step 10/10 : RUN apt -y install ./sdrangel-107-master/sdrangel_4.14.7-1_amd64.deb
 ---> Running in 1ea6a06421d4

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 sdrangel : Depends: libavcodec57 (>= 7:3.4.6) but it is not installable or
                     libavcodec-extra57 (>= 7:3.4.6) but it is not installable
            Depends: libavformat57 (>= 7:3.4.6) but it is not installable
            Depends: libavutil55 (>= 7:3.4.6) but it is not installable
            Depends: libopencv-core3.2 but it is not installable
            Depends: libopencv-imgcodecs3.2 but it is not installable
            Depends: libopencv-imgproc3.2 but it is not installable
            Depends: libopencv-videoio3.2 but it is not installable
            Depends: libswresample2 (>= 7:3.4.6) but it is not installable
            Depends: libswscale4 (>= 7:3.4.6) but it is not installable
E: Unable to correct problems, you have held broken packages.
The command '/bin/sh -c apt -y install ./sdrangel-107-master/sdrangel_4.14.7-1_amd64.deb' returned a non-zero code: 100
exit status 100
```

f4exb, 2020-06-20:
Of course... https://github.com/f4exb/sdrangel/releases/download/v4.14.7/sdrangel-107-master.tar.gz is a 18.04 version. Try: https://github.com/f4exb/sdrangel/releases/download/v4.14.9/sdrangel-124-master-20.04.tar.gz

vk6flab, 2020-06-20:
When this  issue was raised on the 20th of March, that release did not exist.

I can also confirm that you type faster than I ;-)

I edited my own Dockerfile with your new release and confirm that it installs properly with 20.04.

f4exb, 2020-06-20:
Now as advertised the latest release is for 20.04 exclusively: https://github.com/f4exb/sdrangel/releases/download/v4.14.10/sdrangel-137-master.tar.gz

alphafox02, 2020-06-20:
Can the 20.04 builds still be used with 18.04 so long as you perform the non standard steps mentioned above? I ask because I include sdrangel in a 18.04 project I work on called DragonOS LTS. Hope you don’t mind me including it.

f4exb, 2020-06-20:
I don't think so because you will encounter the same dependency issues as in here: https://github.com/f4exb/sdrangel/issues/524#issuecomment-646970606

alphafox02, 2020-06-20:
Okay thank you. I’ll make sure I back everything up before attempting. I think I have it on 4.14.8 now.

f4exb, 2020-06-20:
v4.14.9 still has both versions.
Now this issue can be closed.

closed 2020-06-20 by f4exb