f4exb/sdrangel #525: DVB-S2 frozing every 2 Seconds

dominik0815, 2020-05-20:
Hey Guys,

I want to use the DVB-S2-Mode for decoding QO-100 Satellite and test own transmissions localy. When I start decoding DVB-S it run really good, but when changing the protocoll to DVB-S2, it freeze the QPSK signal diagram every two seconds.

PC-Specs: Win10(64), AMD Ryzen 7, 32GB Ram, Vers. 4.14.7

It would be really nice to use SDR to receive Ham Satellite on PC.

Greetings, Dominik
[image]

f4exb, 2020-05-20:
DVB-S2 is not workable for QO-100 as you need at least 15dB and preferably 20dB S/N to make it work.  

This behavior is normal and due to the processing that produces constellation points at a relatively slow rate particularly if it does not find synchronization which is the case with QO-100.

dominik0815, 2020-05-21:
Sure, the Signal is to low. My Problem is, that the decoding does´nt run very smothely when the signal is strong enough.
[image]

f4exb, 2020-06-20:
Question answered

closed 2020-06-20 by f4exb