f4exb/sdrangel #528: Funcube Dongle Pro+

loewal, 2020-05-29:
Qthid and Audacity were the hints...but helas, until now I haven't been able to let SDRangel (4 and 5) recognize my adapter.
Can anyone please?

f4exb, 2020-05-30:
I suppose you run on Mac O/S and Mac O/S is not officially supported so please look for help from other Mac O/S users in the discussion group rather than opening an issue on Github: https://groups.io/g/sdrangel

loewal, 2020-05-30:
Yes and no!
I'am running on Macos ánd Lubuntu.
The same issue. Not possible to recognize Funcube Donge Pro+.
And yes, qthid works and sees it and so does Audacity.
So......what else can I do.
Besides that, found nothing on the fora.

f4exb, 2020-07-09:
Sorry but this works. Hence closed.

closed 2020-07-09 by f4exb