f4exb/sdrangel #529: HELLO I WANT TO KNOW THAT MAKES THE FUNCTION

gaspar1987, 2020-05-29:
[image]
HELLO I WANT TO KNOW THAT MAKES THE FUNCTION

f4exb, 2020-05-30:
Please use sensible case in the title and express yourself correctly to be understandable

gaspar1987, 2020-05-30:
Hello I wanted to know if it worked to enlarge the frames or reduce them to
any small little cuckoo I could set it to 200 s I do not know if it's
seconds or stuff I do not understand and I would like to know what is used
to listen to trams or listen to analog.n

f4exb, 2020-05-30:
Sorry this is still not clear...
  - what "frames" are you talking about?
  - what the heck is that "little cuckoo"?!
  - "S/s" stands for samples per second this is a sampling rate unit compatible with Hertz (Hz) but more expressive in this case. If you hover on the control the tooltip will give you more details (same thing on all controls on the GUI).
  - " I would like to know what is used to listen to trams or listen to analog.n": what does this mean really?

If your question is "Can I get a lower sampling rate than 230 kS/s on the RTL-SDR" then the answer is "no" because this is a hardware limitation.

f4exb, 2020-06-03:
This can be closed

closed 2020-06-03 by f4exb