f4exb/sdrangel #531: Allow more than 10GHz range on tuning dials

v0l, 2020-06-02:
Hi,

Could you please allow for more than 10GHz tuning range on channels and also on center frequency?

If the sample rate is > 20MS/s its not possible to tune lower or higher than 10GHz on any channels.

This is obviously not high priority but it would be nice to have.

[image]

f4exb, 2020-06-02:
Well... it depends on the demodulator (or modulator). For narrowband FM as shown on the screenshot it is not sensible to use it on such a large baseband in the first place. This could be considered for channels that may use relatively large bandwidths like Wide FM, Broadcast FM, ATV, DATV.

f4exb, 2020-06-05:
So... OK for wideband channel plugins.

f4exb, 2020-07-01:
Implemented in v4.14.14 and v5.7.6

closed 2020-07-01 by f4exb