f4exb/sdrangel #532: feature request: squelch-based recording

febs, 2020-06-04:
I think it would be super useful to record a slice of the spectrum _only_ when its signal is above a certain threshold.

Would that be feasible? Thanks!

f4exb, 2020-06-05:
I've been thinking of it. It makes sens but in fact it is much more complex than it seems because the way SDRangel is organized in multiple channels. Which channel is going to trigger the recording then? And is it the channel activity that triggers the recording or a local increase in the spectrum PSD within the whole baseband or a part of it?

Currently on the v5 branch an ancillary Python script `scriptsapi/superscanner.py` is being developed using both the REST API for control and the websocket spectrum server to sense the spectrum in real time. I've been playing with it for a while and it allows the tracking of multiple channels simultaneously doing the same job as having multiple fixed channels running in parallel but saving a bit of processing. 

So I'd rather go this way augmenting the functionality of this script or creating another specialized one than developing something in the core of SDRangel code.  

Also the present monolithic `.sdriq` file recording system is not really suitable because it makes more sense to be able to record in chunks within the same file. Enhancing the recording capability supporting the SigMF  specification (or an extension of it) is on-going. See: #380 

At present the recording facility is implemented at device level thus covering the whole baseband. It could be interesting also to have a channel dedicated to recording thus capable of recording a limited part of the baseband centered anywhere on this baseband. No issue is opened for this yet but this is also in the back of my mind.

febs, 2020-06-05:
Thank you very much for the throughtful answer. 

In my imagination the recording would be part of each demodulator actually; and maybe allow to record _either_ the raw input or the decoded audio, as well. That'd be super cool. Just like letting a fishing rod in a pond with a bell attached. ;-)

f4exb, 2020-06-20:
> At present the recording facility is implemented at device level thus covering the whole baseband. It could be interesting also to have a channel dedicated to recording thus capable of recording a [...]

Well this is the purpose of these tickets:
  - #543 
  - #544

f4exb, 2020-07-17:
Both new plugins will implement a so called "spectrum squelch" that is based on FFT analysis. If any peak in the spectrum goes above a certain (squelch) level then recording is triggered and stopped when there is no peak above this level anymore (regardless of frequency bin). I think this is more convenient than checking the global channel power. Plugins will be assorted with a spectrum display so the channel sink spectrum can be visualized. There will be also a configurable pre and post recording period so that the signal of interest is fully represented in the recording.

Hence implementation of the two above mentioned tickets will allow to close this one too.

For finer spectrum power detection like defining exclusion and inclusion bands one would have to resort to REST API. Since transmission of spectrum data via websockets is available in v5 only this will be a v5 only feature.

f4exb, 2020-07-17:
> In my imagination the recording would be part of each demodulator actually

Maybe for audio but not for IQ that is demodulation agnostic. This would not make sense.

f4exb, 2020-08-09:
Done in v4.15.0 and v5.9.0

closed 2020-08-09 by f4exb