f4exb/sdrangel #533: Bias-T for RTL-SDR v3

febs, 2020-06-05:
Here's another feature request.

The most modern RTL-SDR (version 3) is capable of providing 4.5 volts "via" bias-tee, which can be enabled "entirely in software" [https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/](https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/)

It would be super cool to be able to toggle it via sdrangel, just as it can be done with the hackrf nowadays.

Thanks,

f4exb, 2020-06-05:
It shouldn't be too difficult provided the function is available in mainstream. It looks like something like this: `rtlsdr_set_bias_tee(dev, enable_biastee);`

febs, 2020-06-27:
Just tested with a multimeter and it works great. Thanks!

closed 2020-06-27 by febs

f4exb, 2020-06-27:
Thanks for confirming I have no v.3 available.