f4exb/sdrangel #537: LimeSDR Mini high LO leakage when using external reference

rfamp, 2020-06-07:
Hi, first of all thank you for the great project it is amazing!

I used the limesdr mini multiple times without an external oscillator and had no problems at all using it.
Now I added a 10MHz external reference clock and selected the reference frequency in sdrangle.
But now I have a rather high LO leakage. The LO is just 20dB lower than the testsignal transmitted using the SSB Modulator.
To compare I used a gnuradio script and the basicTX example from the limesuite repository. With both I got down to around 45-50 dBc LO leakage which fits the 50dBc stated in the LimeSDRs FAQ.

I went through the limesdrsink plugin but I could find anything useful which explains the high LO leakage.
Regards

f4exb, 2020-06-08:
LO leakage is a hardware issue. I do not see the connection with the software.

rfamp, 2020-06-08:
The Lo leakage is usually around -50dB but when using an external reference it is just -20dB when using sdrangel. When using gnuradio or a example program from the Limesuite repo the LO leakage is the same ( -50dB) regardless if the internal clock or an external is used.

closed 2021-02-14 by f4exb