f4exb/sdrangel #540: private

gaspar1987, 2020-06-16:
Hello, I wanted to know how to make the p.
Go active.
The private one.
From.
Small locker. If it had to put it in 99 to listen to the sound perfectly in.
But without the sound.
Fifty. I don't know, I'd like a little help.
I'll take the picture for you.
To explain my body.
yes, I don't know what to do.
[image]

f4exb, 2020-06-16:
This is the basic privacy index from 0 (no BP active) to 255. Just flip through until you hear something readable or (better as you are not supposed to break privacy ;-) ) set it to the value given by the service.

closed 2020-07-09 by f4exb

gaspar1987, 2020-07-16:
J'ai testé avec mon petit walki mais ça ne font pas le mode manuel le son
est un peu n'est pas clair. C'est pour cela que je ne comprends pas
pourquoi il est en mode Manuel le BP au lieu aussi d'être automatique je ne
comprends pas.

markcarter00, 2023-11-16:
how can i get sdrangel with bp softwear for windows download my emaill is myscanner22@hotmail,com