f4exb/sdrangel #541: DMR BP

gaspar1987, 2020-06-16:
Bonjour je suis la rechercher un décode basilic privatif ; j ai vu vous logiciel est bien il marque bouton le auto scan BP est pour le chercher rapide ,je trouve que logiciel peut  avoir de  plus décode numérique en Norne DMR . J ai sur trame DMR CA marche bien ,mais il faut amorale sdr .
Pour la 4.19.9 !

f4exb, 2020-06-16:
Scanning BP indexes is not and will not be supported. You can access BP index setting via REST API thus implementing scanning via an external program (ex: Python script) is possible. 

Good luck to find an automated way to check if the BP index is correct or not. I suppose the only way is to play a recorded sequence in a loop and scan indexes until you hear something readable.

f4exb, 2020-07-09:
Let's be clear: I will not actively promote illegal activities. This issue can be closed.

closed 2020-07-09 by f4exb

gaspar1987, 2020-07-09:
Bonjour j'ai utilisé le format informatique depuis sa création je ne crois
pas avoir fait des bêtises dessus mets c'est celui-ci que j'utilise le plus
que près des sdrshap . Je sais qu'il faut respecter la norme des protocoles
de radio que l'on ne peut pas faire le stage paraKito c'est je sais que
c'est interdit.

f4exb, 2020-07-09:
Ce qui est illégal c'est écouter des conversations qui ne vous sont pas destinées ou non pulbiques. Je pense ne pas trop m'avancer en pensant que c'est ce que vous cherchez à faire avec votre "auto scan".

gaspar1987, 2020-07-10:
Bonjour Monsieur en fait j'utilise un talkie-walkie dmr  avec de la crypto
bP 64bis en mode privatif pour éviter que les pirates . Merci vos logiciel
me plaît c'est pour ça que je les garde sur un disque dur voilà.

f4exb, 2020-07-12:
Si vous connaissez le code alors pas besoin de faire défiler les codes de manière automatique. De plus il est très difficile de réaliser ce dispositif qui doit estimer le bon code lorsque la voix est intelligible. Joli sujet pour une intelligence artificielle mais je ne vais pas me lancer là dedans.

Petit rappel: le BP du DMR est sur 8 bits et non sur 64 bits.

gaspar1987, 2020-07-13:
Bonjour Monsieur j'ai réussi à écouter grâce au code bp en manuel les Motorola en dmr.

gaspar1987, 2020-07-13:
Par contre je voudrais que vous mettez plus de norme comme la norme dPMR si vous réussi à la mettre sur le DSd version 1.9. où la norme dPMR 446 .

f4exb, 2020-07-13:
DSDcc et donc SDRangel implémentent les normes DMR, dStar, YSF, NXDN et dPMR. A base de codec AMBE mais pas avec les codecs chinois (dPMR pas cher...)