f4exb/sdrangel #545: Build broken - DSDcc::DSDDecoder’ has no member named ‘setDMRBasicPrivacyKey

dslotter, 2020-06-18:
One of the recent commits breaks the build. I built this a week ago from source without issue. However, there have been new commits, especially around DMR functionality, and I see a DMR-related compile failure below. Did someone forget to commit a file?

[ 35%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/sdrgui_autogen/mocs_compilation.cpp.o
In file included from /opt/build/sdrangel/plugins/channelrx/demoddsd/dsddemod.h:31,
                 from /opt/build/sdrangel/build/pluginssrv/channelrx/demoddsd/demoddsdsrv_autogen/EWIEGA46WW/moc_dsddemod.cpp:9,
                 from /opt/build/sdrangel/build/pluginssrv/channelrx/demoddsd/demoddsdsrv_autogen/mocs_compilation.cpp:2:
/opt/build/sdrangel/plugins/channelrx/demoddsd/dsddecoder.h: In member function ‘void DSDDecoder::setDMRBasicPrivacyKey(unsigned char)’:
/opt/build/sdrangel/plugins/channelrx/demoddsd/dsddecoder.h:53:63: error: ‘class DSDcc::DSDDecoder’ has no member named ‘setDMRBasicPrivacyKey’
     void setDMRBasicPrivacyKey(unsigned char key) { m_decoder.setDMRBasicPrivacyKey(key); }
                                                               ^~~~~~~~~~~~~~~~~~~~~
make[2]: *** [pluginssrv/channelrx/demoddsd/CMakeFiles/demoddsdsrv.dir/build.make:66: pluginssrv/channelrx/demoddsd/CMakeFiles/demoddsdsrv.dir/demoddsdsrv_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:8180: pluginssrv/channelrx/demoddsd/CMakeFiles/demoddsdsrv.dir/all] Error 2

dslotter, 2020-06-18:
Using: git reset --hard "8ad440f893cc110d7df493ed8c49c4c70e5db94a" I can build successfully. I can not build with source files after that checkpoint.

Tagging @f4exb since the only commits after that are his as of 2020-06-18 @ 14:00 UTC.

dslotter, 2020-06-21:
I'm following the compilation instructions at https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux which indicates to 'git reset --hard "v1.8.7"' on DSDcc but no such restriction on sdrangel. Perhaps the compilation instructions are no longer valid and need to be updated?

f4exb, 2020-07-09:
Closed as unreproducible

closed 2020-07-09 by f4exb

dslotter, 2020-07-09:
No response to my comment, "I'm following the compilation instructions at https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux which indicates to 'git reset --hard "v1.8.7"' on DSDcc but no such restriction on sdrangel. Perhaps the compilation instructions are no longer valid and need to be updated?"

That's how you reproduce the issue... Or should I open up a defect against the documentation instead?

f4exb, 2020-07-09:
I compile this software every day and several times a day moreover compilation is also run successfully on Appveyor. The dsdcc library version used in all cases is v1.8.7 so this works.

dslotter, 2020-07-09:
Do you compile the software according to the instructions at https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux ? I bet you don't, because if you did, you would see the reproducible issue I reported. So again, should I report this as a documentation defect?

If you still don't believe me, then let me know and I'll schedule a videoconferencing call with you so I can reproduce the issue right in front of you.

f4exb, 2020-07-09:
Your error is `DSDcc::DSDDecoder’ has no member named ‘setDMRBasicPrivacyKey` right? this method is new to v1.8.7 so by setting the dsdcc library to this version with the `git reset` you get this method in the headers. This is a clear proof that the version used in your setup is not v1.8.7. Instructions do not supersede judgment so please try to understand what you are doing.

dslotter, 2020-07-09:
The change to DSDcc causing the issue was **_not_** made in tag 1.8.7. It was made **_afterwards_**..

And please don't talk down to me -- try to understand how you might feel being talked down to yourself.

f4exb, 2020-07-09:
You keep affirming things against evidence: https://github.com/f4exb/dsdcc/blob/master/dsd_decoder.h#L260 and it starts to be irritating.

dslotter, 2020-07-09:
What's irritating is a developer not actually reading and comprehending what I'm expressing in clear language, and talking down to a person who is trying to help fix a problem.

Here, run these tasks below in Ansible yourself to see what I'm talking about.

Pay special attention to this part towards the end, which is what I had to do to make the build succeed:

  # Adding this since a recent commit from @f4exb breaks the build.
  - name: Git Reset sdrangel CMakeFiles
    command: **git reset --hard "8ad440f893cc110d7df493ed8c49c4c70e5db94a"**
    args:
      chdir: /opt/build/sdrangel

-----

  - name: Git clone latest CM265cc sources
    git:
      repo: https://github.com/f4exb/cm256cc.git
      dest: /opt/build/cm256cc

  - name: Git Reset CM265cc CMakeFiles
    command: git reset --hard c0e92b92aca3d1d36c990b642b937c64d363c559
    args:
      chdir: /opt/build/cm256cc

  - name: Create directory cm256cc/build
    file:
      path: /opt/build/cm256cc/build
      state: directory

  - name: Build CM265cc CMakeFiles
    command: cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/cm256cc ..
    args:
      chdir: /opt/build/cm256cc/build

  - name: Build CM265cc
    command: make
    args:
      chdir: /opt/build/cm256cc/build

  - name: Install CM265cc
    become: yes
    command: make install
    args:
      chdir: /opt/build/cm256cc/build

  - name: Remove CM265cc build directory
    file:
      path: /opt/build/cm256cc
      state: absent

#
# MBElib
#

  - name: Git clone latest MBElib sources
    git:
      repo: https://github.com/szechyjs/mbelib.git
      dest: /opt/build/mbelib

  - name: Git Reset MBElib CMakeFiles
    command: git reset --hard 9a04ed5c78176a9965f3d43f7aa1b1f5330e771f
    args:
      chdir: /opt/build/mbelib

  - name: Create directory mbelib/build
    file:
      path: /opt/build/mbelib/build
      state: directory

  - name: Build MBElib CMakeFiles
[... 585 lines cut ...]
  - name: Create directory sdrangel/build
    file:
      path: /opt/build/sdrangel/build
      state: directory

  - name: Build SDRAngel CMakeFiles
    command: cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DMIRISDR_DIR=/opt/install/libmirisdr -DAIRSPY_DIR=/opt/install/libairspy -DAIRSPYHF_DIR=/opt/install/libairspyhf -DBLADERF_DIR=/opt/install/libbladeRF -DHACKRF_DIR=/opt/install/libhackrf -DRTLSDR_DIR=/opt/install/librtlsdr -DLIMESUITE_DIR=/opt/install/LimeSuite -DIIO_DIR=/opt/install/libiio -DPERSEUS_DIR=/opt/install/libperseus -DXTRX_DIR=/opt/install/xtrx-images -DSOAPYSDR_DIR=/usr/local/lib/SoapySDR -DCM256CC_DIR=/opt/install/cm256cc -DDSDCC_DIR=/opt/install/dsdcc -DSERIALDV_DIR=/opt/install/serialDV -DMBE_DIR=/opt/install/mbelib -DCODEC2_DIR=/opt/install/codec2 -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
    args:
      chdir: /opt/build/sdrangel/build

  - name: Build SDRAngel
    command: make
    args:
      chdir: /opt/build/sdrangel/build

  - name: Install SDRAngel
    become: yes
    command: make install
    args:
      chdir: /opt/build/sdrangel/build

  - name: Remove SDRAngel build directory
    file:
      path: /opt/build/sdrangel
      state: absent

f4exb, 2020-07-09:
Using Ansible is a good idea and I like it but the yaml looks nicer in this board with the triple quoting. There is also a Docker file in sdrangel-docker project that can be used as a reference but an Ansible playbook is probably more readable.  

However you should proof read your playbook because as one can see it is checking out DSDcc  v1.8.6 not v1.8.7:
```yaml
  - name: Git Reset DSDcc CMakeFiles
    command: git reset --hard "v1.8.6"
    args:
      chdir: /opt/build/dsdcc
```
Which in turn prevents you to use the code in SDRangel that implements this new functionnality of DSDcc past commit `8ad440f893cc110d7df493ed8c49c4c70e5db94a`. So be nice stop kidding and use DSDcc v1.8.7. However thanks for the Ansible idea. At least this makes things clear. 

You should proof read the playbook a bit more because there are some inconsistencies like:
```yaml
  - name: Build RTL-SDR CMakeFiles
    command: cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/libmirisdr-4 ..
    args:
      chdir: /opt/build/librtlsdr/build
```
Of course you can be lucky and have no overlaps between libraries but this is more consistent:
```yaml
  - name: Build RTL-SDR CMakeFiles
    command: cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/librtlsdr ..
    args:
      chdir: /opt/build/librtlsdr/build

dslotter, 2020-07-10:
Thanks for the second set of eyeballs on the Ansible tasks. I made that minor suggested change.

And I see now that you edited the wiki page called "Compile from source in Linux" 22 days ago -- the same exact time range I reported that the build was broken with DSDcc 1.8.6. You changed that page to reflect 1.8.7 just after I noted the build breakage.

There's nothing else to do now. You may consider this matter closed.

f4exb, 2020-07-11:
This is the trouble with hand written documentation. It is sometimes not up to date or not exactly in sync. We started to be in sync when you wrote this Ansible playbook. I have used Ansible at work recently but did not make the connection with compiling SDRangel. Using Ansible is actually a very good idea to make code and documentation match as it becomes part of the codebase it becomes also part of the commits. It is also very easy to launch it in a VM and see if there is any mistake. I may retain this idea for the future thanks!