f4exb/sdrangel #547: channel analyzer broken?

pe1iee, 2020-06-19:
Using the windows 4.14.9 version I noticed that the channel analyzer shows the channel spectrum but no signal shows in the channel scope window, just a steady line. 
Am I missing some extra settings here?  I used 4.0.7 before this, where it showed the scope signal directly.  Please advice. 
Peter, PE1IEE

f4exb, 2020-06-19:
Indeed suspicious. Maybe stg around this commit: https://github.com/f4exb/sdrangel/commit/9945c6a8f8edc527faa6c4b6c043efc92ecadcab

f4exb, 2020-06-20:
Fixed in v4.14.10

closed 2020-06-20 by f4exb

pe1iee, 2020-06-20:
Wow! Thanks a lot for the super fast response!
I like this plugin very much and use it often so this is super.
SDRangel is by far the best tool for hams.
73 Peter