f4exb/sdrangel #551: Is Sdrangel tested against xtrx PCIe or xtrx USB ?

OhSoGood, 2020-06-25:
Hi,

I was wondering on what platform XTRX owners were testing : with the USB version and/or the PCIe version? A friends and I have the XTRX-CS on PCIe (not the XTRX Pro) and whatever we do (under gnuradio, soapy, sdrangel, we always get flat spectrum, even when the sdr is connected by SMA cable to a signal generator. With other SDR (blade or usrp), everyhting is fine. So, could it be that none of you tested the same combination as us - ie XTRX-CS on PCIe (which looks unlikely), or that our xtrx are simply dead ? Or would you see blattant mistake in our settings of SDRAngel?

We have tried to ask Fairwaves, but these unfair guys never answer. Worst, the co-founder and main software/firmware developer has just left the company.

Here is what we get under Ubuntu 20.04, with a version compiled as of today:
[image]

f4exb, 2020-06-26:
I can confirm that I only test against XTRX USB. I don't have a PCIe and not aware of differences in handling. Apparently there might be...

f4exb, 2020-07-09:
I think this answers the question. I have not the appropriate hardware to develop a PCIe version if this is any different. This cannot be carried on further.

closed 2020-07-09 by f4exb