f4exb/sdrangel #554: Initialisation of AM Demod consumes 2.7 GB of RAM

Vort, 2020-06-29:
Class `AMDemodSink` contains field `m_audioSampleRate`, which is not initialised at constructor `AMDemodSink::AMDemodSink()`.
Because of this, when `AMDemodBaseband` is created inside `AMDemod::AMDemod(DeviceAPI *deviceAPI)`, `AMDemodSink::applySettings(const AMDemodSettings& settings, bool force)` function executes line:
https://github.com/f4exb/sdrangel/blob/9c05329bb3dc12b828f065e7aa7d00ed84b3de1f/plugins/channelrx/demodam/amdemodsink.cpp#L286
And if accidently `m_audioSampleRate` contains large value, huge array becomes allocated:
[image]

f4exb, 2020-06-30:
That's right thanks for pointing out. It is in fact used all over the place. It will eventually be set through the rest of the application but it can create a lot of intermediate havoc. It can be initialized with an arbitrary value like 48000.

f4exb, 2020-06-30:
I will also chase this bug in other plugins having audio that may be affected in a similar way.

febs, 2020-07-01:
I guess this is why SDRangel hang indefinitely when adding an audio plugin in the Docker image.

f4exb, 2020-07-01:
It depends... if you look at the above commit you will see that not all plugins are affected.

commit 41a837c2 (2020-07-01): Initialize m_audioSampleRate in constructors where missing. Fixes #554
commit 563c509c (2020-07-01): Initialize m_audioSampleRate in constructors where missing. Fixes #554
closed 2020-07-01 by f4exb