f4exb/sdrangel #555: DC spurs with RTL-SDR V3

febs, 2020-07-01:
I am using SDRangel 5.7.3 on Ubuntu Linux 19.10.

Using RTL-SDR V3 dongle, acquiring 2.4M samples per seconds **and using a decimation of 16**.
I get this sequence of DC spurs. I'm pretty sure it's DC because it's there no matter where I tune in.
[image]

If I disable the "DC" correction option the center peak also appears:
[image]

This does NOT happen when using the no-mod direct sampling feature of the dongle, so I guess it's something introduced by the tuner chip somehow. 

It looks like the DC filter is filtering the main spike and not its harmonics?

Thanks,

f4exb, 2020-07-01:
That's right the purpose of the DC filter is to remove DC. And you can see that it does its job pretty well. The rest is supposed to be a useful signal. You have to take appropriate measures to remove unwanted signals the software cannot do anything for you there.

And errm... are you seriously expecting harmonics of a DC signal? (n times zero is still zero)

febs, 2020-07-01:
The "harmonics" thing was just a guess, apologies for it being nonsense.

What I can tell for sure, though, is that the rest is _not_ an useful signal. No matter where I tune, the FTT is identical, from 28 Mhz
[image]
all the way up to 1.9 Ghz
[image]

that appears anywhere I tune in between, too.

**It may be worthy to add that this happens only with the Automatic Gain Control enabled.**

f4exb, 2020-07-01:
This looks like pretty much a hardware issue than a software issue. I am using RTL-SDRs for QO-100 reception and a few remote receivers and everything works fine. These are not v3s but NooElec TCXO SMA connectorized versions. I there was something so enormous I would have noticed it. So you'd better look for hardware issues and question your reception chain.

Recently you have requested the ability to use the bias tee so I suppose you have a LNA in the way. If the gain of the LNA is too high and the signal coming from the antenna is not filtered it is quite possible that you have a strong signal outside the passband. Normally switching on the AGC would mitigate overload issues and not worsen it but this works only up to a certain point. Maybe this unwanted signal puts the AGC in an undesirable state while switching it off just blinds the RTL-SDR input stage and then nothing happens. 

Maybe you are new to SDR. A golden rule that not only applies to SDR but becomes critical with it is to filter the signal right after the antenna. This may not be necessary at very high frequencies with a narrow beam antenna (i.e. satellite reception) but today there are so many strong signals all over up to a few GHz that it becomes an absolute necessity.

febs, 2020-07-01:
I actually don't have a LNA yet. I ordered it just after you implemented the GUI checkbox to enable it, otherwise I wouldn't have been able to use it. :-)
So it must be the dongle itself.
I will experiment with gnuradio and see what I get in it. 

Thank you very much,

febs, 2020-07-01:
Btw, I just wanted to add that it is something noticeable only when "zooming in" a lot by using decimation up to 16.  With "1" it is negligible.

febs, 2020-07-29:
I accidentally got another RTL-SDR v3 dongle (I ordered their LNA but I got that instead by accident LOL). The behavior is the same. 
If you're interested I can send it this spare dongle to you (at no charge of course) so that you can fiddle with it and see how it behaves to you. Or just as a donation for your work if you thing you might need it for any purpose.

f4exb, 2020-08-09:
That's nice of you but I think after re-reading the complete thread that the issue is not with software. In particular you said that you observe these spurs all over the frequency range which means that these spurs appear at the IF of the RTL-SDR HF chain. Now I have never observed this with the other flavors of RTL-SDRs. The singularity of v.3 is to have bias-T it is possible that this introduces some noise due to the voltage regulator circuit.

Vort, 2020-08-12:
You can plug resistor instead of antenna and look if glitch will go away.
Also you can record that glitch and share .sdriq file.
Maybe measurements of frequency can give some ideaes about origins of glitch signal.

f4exb, 2020-08-13:
That's right connecting a SMA load at the place of the antenna will rule out signals coming in. It is probably something internal but let's rule out this case.

If this was decimation as the same decimation routine is used everywhere (half band decimators) the same issue would show up for other devices. Can you try with somethnig else than a RTL-SDR v3?

Also can you try with another flavor of RTL-SDR i.e. the cheap "blue one" or one from NESDR? I have used these a lot without noticing a similar issue so chances are it comes from the internals of v3 that are indeed different in order to maintain the bias-T.

Vort, 2020-08-13:
It is also worth trying different PC.
Since bad signal can come from PC.

febs, 2020-08-24:
> I think that computer (or its position) differences are more likely cause
> than OS differences.

The computers are in the very same positions (on the same table). BTW I
noticed those spurs in the system where they happen in several different
places even 300+ km away from each other.
I will try hard to build sdrangel on linux on that second system and see
what happens, I'm very curious to dig deeper into this.

Thanks for your inputs.

f4exb, 2020-11-03:
No follow up in SDRangel code. This issue can be closed.

closed 2020-11-03 by f4exb