f4exb/sdrangel #556: Moving of iterator past the end in ScopeVis::feed

Vort, 2020-07-01:
When I try to use scope with debug build of sdrangel, error pops out:
`_DEBUG_ERROR("vector iterator + offset out of range");`
It comes from this line:
https://github.com/f4exb/sdrangel/blob/8794021940f8e23fcd88eb4e43788cd78c6d7b30/sdrgui/dsp/scopevis.cpp#L248
From the [answer on stackoverflow](https://stackoverflow.com/questions/39820041/is-it-allowed-to-increment-an-end-iterator) it looks like incrementing iterator past end is undefined behaviour.
Maybe it will be better to replace this line with
`if (end - begin < m_traceSize) // buffer smaller than trace size`
?

f4exb, 2020-07-01:
I have not encountered an issue with this but this may rely on the compiler being smarter than the programmer which is not very safe in general. Indeed `if (end - begin < m_traceSize)` makes more sense.

f4exb, 2020-07-06:
Implemented in v5.7.8 and v4.14.15

closed 2020-07-06 by f4exb