f4exb/sdrangel #559: Adding MIMO device causes crash

v0l, 2020-07-07:
**Version:** tags/v5.7.8-0-gb486b1dda

```
2020-07-08 04:01:02.488 (D) MainWindow::MainWindow: end
2020-07-08 04:01:02.542 (D) GLSpectrum::initializeGL: context:  major:  4  minor:  6  ES:  no
2020-07-08 04:01:04.808 (D) DSPDeviceMIMOEngine::start
2020-07-08 04:01:04.808 (D) DSPDeviceMIMOEngine::run
2020-07-08 04:01:04.841 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_linear:  false  m_ssb:  false  m_usb:  true  m_wsSpectrumAddress:  "127.0.0.1"  m_wsSpectrumPort:  8887  force:  true
2020-07-08 04:01:04.841 (D) FFTFactory::getEngine: create engine: 1 FFT fwd size: 1024
2020-07-08 04:01:04.841 (D) FFTEngine::create: using FFTW engine
2020-07-08 04:01:04.841 (D) FFTWEngine::configure: no FFTW wisdom file
2020-07-08 04:01:04.841 (D) FFT: creating FFTW plan (n=1024,forward) took 0 ms
2020-07-08 04:01:04.841 (D) GLSpectrum::connectTimer
2020-07-08 04:01:04.844 (D) GLSpectrumGUI::on_averaging_currentIndexChanged: log
2020-07-08 04:01:04.844 (D) GLSpectrumGUI::on_fftSize_currentIndexChanged: 3
2020-07-08 04:01:04.844 (D) DSPDeviceMIMOEngine::addSpectrumSink:  SpectrumVis
2020-07-08 04:01:04.844 (D) DSPDeviceMIMOEngine::handleSynchronousMessages:  DSPDeviceMIMOEngine::AddSpectrumSink
2020-07-08 04:01:04.888 (D) DSPDeviceMIMOEngine::setMIMO
2020-07-08 04:01:04.889 (D) DSPDeviceMIMOEngine::handleSynchronousMessages:  DSPDeviceMIMOEngine::SetSampleMIMO
2020-07-08 04:01:04.889 (D) DSPDeviceMIMOEngine::handleSetMIMO: asynchronous sources set TestMI channel 0
2020-07-08 04:01:04.889 (D) DSPDeviceMIMOEngine::handleSetMIMO: asynchronous sources set TestMI channel 1
2020-07-08 04:01:04.889 (D) TestMIGui::TestMIGui
2020-07-08 04:01:04.890 (D) ValueDialZ::formatText: value: 0
2020-07-08 04:01:04.890 (D) DSPDeviceSinkEngine::setSpectrumSinkInput:   sourceElseSink:  true  index:  0
2020-07-08 04:01:04.890 (D) DSPDeviceMIMOEngine::handleSynchronousMessages:  DSPDeviceMIMOEngine::SetSpectrumSinkInput
2020-07-08 04:01:04.890 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrum::MsgReportSampleRate
2020-07-08 04:01:04.890 (D) ValueDialZ::formatText: value: 0
2020-07-08 04:01:04.890 (D) ValueDialZ::formatText: value: 0
2020-07-08 04:01:04.890 (D) TestMI::applySettings: common:   m_fileRecordName:  ""  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0
2020-07-08 04:01:04.890 (D) TestMI::applySettings: stream # 0 :   m_centerFrequency:  435000000  m_frequencyShift:  0  m_sampleRate:  768000  m_log2Decim:  4  m_fcPos:  2  m_amplitudeBits:  127  m_sampleSizeIndex:  0  m_autoCorrOptions:  0  m_dcFactor:  0  m_iFactor:  0  m_qFactor:  0  m_phaseImbalance:  0  m_modulation:  0  m_amModulation:  50  m_fmDeviation:  50  m_modulationTone:  44
2020-07-08 04:01:04.890 (D) DSPDeviceMIMOEngine::configureCorrections
2020-07-08 04:01:04.890 (D) DeviceSampleStatic::calculateSourceDeviceCenterFrequency:  frequencyShiftScheme:  0  desired center freq:  435000000  Hz  device center freq:  435000000  Hz  device sample rate:  768000 S/s  Actual sample rate:  48000 S/s  center freq position code:  2  image frequency:  435000000 Hz
Segmentation fault (core dumped)
```

**Call Stack:**
```
libmimotestmi.so!TestMI::applySettings(TestMI * const this, const TestMISettings & settings, bool force) (/opt/build/sdrangel/plugins/samplemimo/testmi/testmi.cpp:481)
libmimotestmi.so!TestMI::init(TestMI * const this) (/opt/build/sdrangel/plugins/samplemimo/testmi/testmi.cpp:95)
libsdrgui.so!MainWindow::addMIMODevice(MainWindow * const this) (/opt/build/sdrangel/sdrgui/mainwindow.cpp:534)
libsdrgui.so!MainWindow::on_action_addMIMODevice_triggered(MainWindow * const this) (/opt/build/sdrangel/sdrgui/mainwindow.cpp:2069)
libsdrgui.so!MainWindow::qt_static_metacall(QObject * _o, QMetaObject::Call _c, int _id, void ** _a) (/opt/build/sdrangel/build/sdrgui/sdrgui_autogen/EWIEGA46WW/moc_mainwindow.cpp:263)
libsdrgui.so!MainWindow::qt_metacall(MainWindow * const this, QMetaObject::Call _c, int _id, void ** _a) (/opt/build/sdrangel/build/sdrgui/sdrgui_autogen/EWIEGA46WW/moc_mainwindow.cpp:302)
libQt5Core.so.5!QMetaObject::activate(QObject*, int, int, void**) (Unknown Source:0)
libQt5Widgets.so.5!QAction::triggered(bool) (Unknown Source:0)
libQt5Widgets.so.5!QAction::activate(QAction::ActionEvent) (Unknown Source:0)
libQt5Widgets.so.5![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libQt5Widgets.so.5!QMenu::mouseReleaseEvent(QMouseEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QWidget::event(QEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QMenu::event(QEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QApplicationPrivate::notify_helper(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QApplication::notify(QObject*, QEvent*) (Unknown Source:0)
libQt5Core.so.5!QCoreApplication::notifyInternal2(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) (Unknown Source:0)
libQt5Widgets.so.5![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libQt5Widgets.so.5!QApplicationPrivate::notify_helper(QObject*, QEvent*) (Unknown Source:0)
libQt5Widgets.so.5!QApplication::notify(QObject*, QEvent*) (Unknown Source:0)
```

f4exb, 2020-07-08:
Which version of SDRangel is this?

v0l, 2020-07-08:
b486b1dda7638b6e6ac9743fe00dc096712da4aa

f4exb, 2020-07-09:
reproduced

f4exb, 2020-07-09:
Since the file recording functionality is going to be redesigned and that it is not of much use for MIMO in the present state I suggest to remove it from all MIMO plugins. Code is already commented out for LimeSDR and XTRX,

f4exb, 2020-07-09:
We need an earlier fix

f4exb, 2020-07-09:
Delivered in v5.7.9

closed 2020-07-09 by f4exb