f4exb/sdrangel #560: ATV Demod: Incorrect initialization of sample rate

Vort, 2020-07-08:
Steps to reproduce:
1. Download and unpack [hacktv_dec5.7z.zip](https://github.com/f4exb/sdrangel/files/4888801/hacktv_dec5.7z.zip) (7-zip + zip).
2. Start sdrangel.
3. Select FileInput source.
4. Select `hacktv_dec5.sdriq` as signal source file.
5. Add ATV Demodulator.
6. Set parameters: AM, HSync, VSync, Inv.
7. Press "Start" button.

<details>
  <summary>Result looks like this:</summary>
  
  [image]
  
</details>

This happens because of incorrect initialization.
At some time samplerate of 48k is fed into ATVDemodBaseband object.
<details>
<summary>Hackfix looks like this:</summary>
  
```patch
diff --git a/plugins/channelrx/demodatv/atvdemodbaseband.cpp b/plugins/channelrx/demodatv/atvdemodbaseband.cpp
index 750fd1a1b..cd1323f73 100644
--- a/plugins/channelrx/demodatv/atvdemodbaseband.cpp
+++ b/plugins/channelrx/demodatv/atvdemodbaseband.cpp
@@ -127,6 +127,8 @@ bool ATVDemodBaseband::handleMessage(const Message& cmd)
     {
         QMutexLocker mutexLocker(&m_mutex);
         DSPSignalNotification& notif = (DSPSignalNotification&) cmd;
+        if (notif.getSampleRate() == 48000) // oh really?
+            return true;
         qDebug() << "ATVDemodBaseband::handleMessage: DSPSignalNotification: basebandSampleRate: " << notif.getSampleRate();
         m_sampleFifo.setSize(SampleSinkFifo::getSizePolicy(notif.getSampleRate()));
         m_channelizer->setBasebandSampleRate(notif.getSampleRate()); 
```

</details>

But, of course, proper solution should be implemented.

f4exb, 2020-07-08:
Maybe a more general issue with the file input because the actual sample rate is not known before the file is opened but at that time maybe the actual sample rate is not propagated correctly. Until the file is open as some objects need to be initialized an arbitrary sort of making sense sample rate must be given.

I think the ATV plugin should work if you use an actual transmitter with the file as a file source.

Vort, 2020-07-08:
ATV plugin initializes correctly if I hit "Start" before adding ATV Demodulator.
I guess the problem is related to this 48k value:
[image]
File is opened, header is read, I thinks FileInput have all needed information to initialize it correctly.
By the way, why there are two sample rates in UI?
Also, AM demod have no such problem (but maybe because AM demod works differently).

f4exb, 2020-07-08:
Indeed this inconsistency in the UI may be the indication of more inconsistencies and leads me to think the root cause is in the File Input plugin.

closed 2020-07-12 by f4exb

f4exb, 2020-07-13:
Done in v4.14.16 and v5.7.10

commit 99586c7b (2020-07-11): ATV demod: fixed initialization issues. Fixes #560
commit 34a8061b (2020-07-12): ATV demod: fixed initialization issues. Fixes #560