f4exb/sdrangel #561: GLSpectrum: freeze state out of sync

Vort, 2020-07-09:
1. Start some signal source.
2. Press freeze button at spectrum display window.
3. Stop signal source.
4. Start signal source again.

After step 2 spectrum is frozen.
But after step 4  it is running again despite button state indicates that it should stay frozen.

It is important that this button should allow to freeze spectrum before signal source is started, because in debug mode spectrum consumes lot of resources, which can be saved by correct implementation of this feature.

<details>
<summary>Here is the proposed patch for fixing of this problem:</summary>

```patch
diff --git a/sdrbase/dsp/spectrumvis.cpp b/sdrbase/dsp/spectrumvis.cpp
index bb617f61f..f88274e96 100644
--- a/sdrbase/dsp/spectrumvis.cpp
+++ b/sdrbase/dsp/spectrumvis.cpp
@@ -39,13 +39,14 @@ MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgConfigureSpectrumVis, Message)
 MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgConfigureScalingFactor, Message)
 MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgConfigureWSpectrumOpenClose, Message)
 MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgConfigureWSpectrum, Message)
-MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgStartStop, Message)
+MESSAGE_CLASS_DEFINITION(SpectrumVis::MsgFreezeUnfreeze, Message)
 
 const Real SpectrumVis::m_mult = (10.0f / log2f(10.0f));
 
 SpectrumVis::SpectrumVis(Real scalef) :
 	BasebandSampleSink(),
     m_running(true),
+    m_isFrozen(false),
 	m_fft(nullptr),
     m_fftEngineSequence(0),
 	m_fftBuffer(MAX_FFT_SIZE),
@@ -321,7 +322,7 @@ void SpectrumVis::feed(const Complex *begin, unsigned int length)
 
 void SpectrumVis::feed(const SampleVector::const_iterator& cbegin, const SampleVector::const_iterator& end, bool positiveOnly)
 {
-    if (!m_running) {
+    if (!m_running || m_isFrozen) {
         return;
     }
 
@@ -670,15 +671,9 @@ bool SpectrumVis::handleMessage(const Message& message)
         handleConfigureWSSpectrum(conf.getAddress(), conf.getPort());
         return true;
     }
-    else if (MsgStartStop::match(message)) {
-        MsgStartStop& cmd = (MsgStartStop&) message;
-
-        if (cmd.getStartStop()) {
-            start();
-        } else {
-            stop();
-        }
-
+    else if (MsgFreezeUnfreeze::match(message)) {
+        MsgFreezeUnfreeze& cmd = (MsgFreezeUnfreeze&) message;
+        m_isFrozen = cmd.getIsFrozen();
         return true;
     }
 	else
diff --git a/sdrbase/dsp/spectrumvis.h b/sdrbase/dsp/spectrumvis.h
index 5e6749826..27894993b 100644
--- a/sdrbase/dsp/spectrumvis.h
+++ b/sdrbase/dsp/spectrumvis.h
@@ -63,22 +63,22 @@ public:
         { }
     };
 
-    class SDRBASE_API MsgStartStop : public Message {
+    class SDRBASE_API MsgFreezeUnfreeze : public Message {
         MESSAGE_CLASS_DECLARATION
 
     public:
-        bool getStartStop() const { return m_startStop; }
+        bool getIsFrozen() const { return m_isFrozen; }
 
-        static MsgStartStop* create(bool startStop) {
-            return new MsgStartStop(startStop);
+        static MsgFreezeUnfreeze* create(bool isFrozen) {
+            return new MsgFreezeUnfreeze(isFrozen);
         }
 
     protected:
-        bool m_startStop;
+        bool m_isFrozen;
 
-        MsgStartStop(bool startStop) :
+        MsgFreezeUnfreeze(bool isFrozen) :
             Message(),
-            m_startStop(startStop)
+            m_isFrozen(isFrozen)
         { }
     };
 
@@ -170,6 +170,7 @@ private:
     };
 
     bool m_running;
+    bool m_isFrozen;
 	FFTEngine* m_fft;
 	FFTWindow m_window;
     unsigned int m_fftEngineSequence;
diff --git a/sdrgui/gui/glspectrumgui.cpp b/sdrgui/gui/glspectrumgui.cpp
index f42bda634..42b3679e9 100644
--- a/sdrgui/gui/glspectrumgui.cpp
+++ b/sdrgui/gui/glspectrumgui.cpp
@@ -342,7 +342,7 @@ void GLSpectrumGUI::on_clearSpectrum_clicked(bool checked)
 
 void GLSpectrumGUI::on_freeze_toggled(bool checked)
 {
-    SpectrumVis::MsgStartStop *msg = SpectrumVis::MsgStartStop::create(!checked);
+    SpectrumVis::MsgFreezeUnfreeze *msg = SpectrumVis::MsgFreezeUnfreeze::create(checked);
     m_spectrumVis->getInputMessageQueue()->push(msg);
 }
  
```

</details>

f4exb, 2020-07-09:
I don't think this is the best solution because if the spectrum stays frozen after restarting the device then the average user will think there is an issue because you can't see anything moving. I'd rather have the start of the device resetting the spectrum state to unfrozen systematically. I agree on the issue though because after restarting the toggle gets in the wrong position.

f4exb, 2020-07-09:
OK I think I got it. What happens is actually what I describe just above so up to this point this is correct. The problem is in fact that the action on the SpectrumVis (start in this case) is not propagated to the UI that remains in the stop (wrong) position.

Vort, 2020-07-10:
And now you have another inconsistency: Press Start, Press Stop -> spectrum is in frozen state, press button to unfreeze it -> and nothing happens.
I think running and frozen should be separate states. And actual activity is determined as running && !frozen. Also if you instead of unfreezing with start make it unfreeze with stop, this will allow to disable spectrum before starting of decoding, which can help save resources.

Vort, 2020-07-10:
> if the spectrum stays frozen after restarting the device then the average user will think there is an issue because you can't see anything moving

He froze it several seconds or minutes ago.
Why he should forget it?

f4exb, 2020-07-10:
I think if the device starts then the spectrum starts this is how I see things and how things are going to be. I am definite on this. Moreover this makes the "freeze" button in sync with the device that feeds it and it is nice like that.

f4exb, 2020-07-10:
> Press Start, Press Stop -> spectrum is in frozen state, press button to unfreeze it -> and nothing happens.

Of course... the device is stopped so where would it get the samples from?

f4exb, 2020-07-13:
Done in v4.14.16 and v5.7.10

closed 2020-07-13 by f4exb