f4exb/sdrangel #564: MIMO Channels not connected

v0l, 2020-07-10:
Hi,

When i add channels to MIMO device they don't seem to be connected, they are not shown on the waterfall and there is no signal on the channels, am i doing something wrong? 

[image]

**Version**: b64ee0c4dae5de038a7ed80d33d549e236e7b95f

f4exb, 2020-07-10:
Can you please also tell us if this is a Lime mini (1 channel but available to a MIMO plugin) or Lime USB (true 2x2 MIMO)? This can lead to different behavior and I have not explored Lime Mini a lot in this situation because of lack of interest to be handled as MIMO (the Rx/Tx sync is not available yet).

v0l, 2020-07-10:
this is for the LimeUSB

f4exb, 2020-07-10:
I am detailing the title because if you try with a MIMO channel (like the Interferometer) I think it works. This is the case of a single stream (normal) channel used with a MIMO device by connectign to one of its channels.

v0l, 2020-07-10:
Yes the Interferometer appears to work ok

f4exb, 2020-07-13:
The issue is that you are adding modulators and you are trying to do Rx thus this has no effect. In the list of channel plugins you have to match the Rx or Tx kind with the side of the device (Rx or Tx) you are using. I realized that because I did the exact same mistake while testing. MIMO is simultaneously Rx and Tx so you have to be careful with this.

This is a different issue but while I was fiddling I realized that removing a channel while the device is running causes a core dump. This is going to be fixed but should not prevent usage if you keep the same channels during the session.

v0l, 2020-07-15:
Ok, so do you want to leave this open for the segfault or?

f4exb, 2020-07-15:
Issues discovered trying your scenario are indeed already documented and fixed so this can be closed. Latest v5.7.11 should work fine.

v0l, 2020-07-15:
Ok, thanks

closed 2020-07-15 by v0l