f4exb/sdrangel #572: Nouvelles versions de Sdrangel

choupette-change, 2020-07-15:
Bonjour,
depuis les 3 ou 4 dernières versions, j'ai des pb de "core dumped" au chargement et/ou au lancement de mes .prex précédents où est le pb que je n'avais pas avant ? est-ce une régression ? Merci pour vos retours.
Ubuntu 20.04 installation par le .deb

gaspar1987, 2020-07-15:
pourvois une nouvelle versions de sdrangel . avec automatique bp  merci

f4exb, 2020-07-17:
In English please

f4exb, 2020-07-17:
Anyway your description in French is too vague and with the little information I have I cannot reproduce this issue. If you do not correct your description in the next few days I will close this issue.

gaspar1987, 2020-07-17:
Hello sir I wanted you to fix it to put it in automatic mode the bp it
lacks a button to make it work because you still hear noise and it does not
exactly correspond to the number I wanted to put it has trouble analyzing.
Thank you

f4exb, 2020-07-17:
> I wanted you to fix it to put it in automatic mode

This is a NO I don't know how I can make it more CLEAR.

> it does not exactly correspond to the number I wanted to put it has trouble analyzing

Implementation is as per Motorola BP standard. If your set does not comply then it may not be able to match.

choupette-change, 2020-07-17:
Hello,
Since the last 3 or 4 versions, I have the following bug "Segment error (core dumped)" if I import and load, in SDRangel, one of my previous files (.prex). Everything worked well before. Thanks for your feedback.
My current installation: Ubuntu 20.04, SDRangel 4.14.17 installed by sdrangel_4.14.17-1_amd64.deb
...................
2020-07-17 13:53:52.314 (D) DSPDeviceSourceEngine::handleInputMessages: message: DSPSignalNotification
2020-07-17 13:53:52.314 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification:  m_sampleRate:  2400000  m_centerFrequency:  96000000
2020-07-17 13:53:52.314 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  SpectrumVis
2020-07-17 13:53:52.314 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  96000000  sampleRate:  2400000
2020-07-17 13:53:52.314 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  FileRecord
2020-07-17 13:53:52.314 (D) FileRecord::handleMessage: DSPSignalNotification: m_inputSampleRate:  2400000  m_centerFrequency:  96000000
2020-07-17 13:53:52.314 (D) DSPDeviceSourceEngine::handleInputMessages: forward message to  BFMDemod
2020-07-17 13:53:52.315 (D) BFMDemod::handleMessage: DSPSignalNotification
2020-07-17 13:53:52.315 (D) DSPDeviceSourceEngine::handleInputMessages: DSPSignalNotification: guiMessageQueue: 0x55a9d10abbe8
Erreur de segmentation (core dumped)
tutu@tutu-V6-531:~$

f4exb, 2020-07-17:
Some changes in plugins may introduce incompatibility with data saved from a previous version particularly if it is far in the past. It looks like something happens in the BFM demod but the console log is not very informative. As you may have noticed the `.prex` file is not binary it is utf-64 and is actually an ASCII representation of binary data. If you paste the content of the `.prex` somewhere (it is a very long line but may be working just here) I could try to load it myself and figure out what is happening.

choupette-change, 2020-07-17:
Hoping to understand your request (Google translation ..), here is the line of one of my .prex files as requested. The other .prex files cause the same problem:
kAABAXABAHACADADBAW42ACBBAGLAAAA/wAAAAD9AAAAAgAAAAAAAAFRAAAC1/wCAAAABPsAAAAaAGkAbgBwAHUAdABWAGkAZQB3AEQAbwBjAGsBAAAAEwAAAS8AAADfAAABL/sAAAAeAGkAbgBwAHUAdABTAGUAbABlAGMAdABEAG8AYwBrAQAAAUgAAACFAAAAhQAAAIX7AAAAJABzAHAAZQBjAHQAcgBhAEQAaQBzAHAAbABhAHkARABvAGMAawEAAAHTAAAAbAAAAGwAAABs/AAAAkUAAAClAAAAkQEAABn6AAAAAAIAAAAC+wAAABQAcAByAGUAcwBlAHQARABvAGMAawEAAAAA/////wAAAHcA////+wAAABgAYwBvAG0AbQBhAG4AZABzAEQAbwBjAGsBAAAAAP////8AAAB3AP///wAAAAIAAAP/AAABofwBAAAAAfsAAAAWAGMAaABhAG4AbgBlAGwARABvAGMAawEAAAFXAAAD/wAAAGgA////AAAD/wAAATAAAAABAAAAAgAAAAEAAAAC/AAAAACABWGQAAEBAAECBAAAAgAAAwEEQAQEwXAAAEAFBEI8AABgBgEAYAcBAGAIAQFgCQEAAAoAYAsBAWAMAQAADQEaAA4BAQAPAShgEAEBABEBOUASBD8rhR8AEwEBABQCA+hgFQEAYAYBAQAHAAAUAQJwGBxzZHJhbmdlbC5zYW1wbGVzb3VyY2UucnRsc2RycBkIMDAwMDAwMDEAGgCAG1KQAAEBAAIBVwADABAEAGAFAQFgBgEAAAcBAgAIAySfAGAJAQBgCgEAYAsBAGAMAQAgDQAQDgBgDwEAYBABAHARCTEyNy4wLjAuMRASAiK4EBMAcBwcc2RyYW5nZWwuc2FtcGxlc291cmNlLmhhY2tyZnAdEDkwOTg2NGM4Mzg4NTY5Y2YAHgCAH1KQAAEBAAEAYAMBARAEAQIABQECYAYBABAHAQwQCAMas/AQCQEWYAoBAWALAQAwDAOYloBgDgEAcA8JMTI3LjAuMC4xEBACIrgQEQBgEgEAIBMAAMgBAXDJFHNkcmFuZ2VsLmNoYW5uZWwuYmZtgcoBBJAAAQEAAQMMNQAAAgEHAAMBDwAEATIABQHEEAcE/wAA/4AIYZAAAQEAAQIIAAACAAADAQNABATCDAAAQAUEQkgAAGAGAQBgBwEAYAgBAGAJAQAACgBgCwEBYAwBAAANARYADgEBAA8BGmAQAQEAEQE+QBIEPyj1wwATAQEAFAIAyGAVAQBgCQEBYAoBAYALKZAAAQEAAQMMNQAQAgT/AAD/cAMSQnJvYWRjYXN0IEZNIERlbW9kAAcAcAwSQnJvYWRjYXN0IEZNIERlbW9kcA0VU3lzdGVtIGRlZmF1bHQgZGV2aWNlYA4BAHAPCTEyNy4wLjAuMRAQAhy7EBEAEBIAABMA

closed 2020-07-17 by choupette-change