f4exb/sdrangel #573: N

choupette-change, 2020-07-17:
oops i'm not used to the github i closed the problem "Nouvelles versions de Sdrangel #572" by mistake. 
Hoping to understand your request (Google translation ..), here is the line of one of my .prex files as requested. The other .prex files cause the same problem:
kAABAXABAHACADADBAW42ACBBAGLAAAA/wAAAAD9AAAAAgAAAAAAAAFRAAAC1/wCAAAABPsAAAAaAGkAbgBwAHUAdABWAGkAZQB3AEQAbwBjAGsBAAAAEwAAAS8AAADfAAABL/sAAAAeAGkAbgBwAHUAdABTAGUAbABlAGMAdABEAG8AYwBrAQAAAUgAAACFAAAAhQAAAIX7AAAAJABzAHAAZQBjAHQAcgBhAEQAaQBzAHAAbABhAHkARABvAGMAawEAAAHTAAAAbAAAAGwAAABs/AAAAkUAAAClAAAAkQEAABn6AAAAAAIAAAAC+wAAABQAcAByAGUAcwBlAHQARABvAGMAawEAAAAA/////wAAAHcA////+wAAABgAYwBvAG0AbQBhAG4AZABzAEQAbwBjAGsBAAAAAP////8AAAB3AP///wAAAAIAAAP/AAABofwBAAAAAfsAAAAWAGMAaABhAG4AbgBlAGwARABvAGMAawEAAAFXAAAD/wAAAGgA////AAAD/wAAATAAAAABAAAAAgAAAAEAAAAC/AAAAACABWGQAAEBAAECBAAAAgAAAwEEQAQEwXAAAEAFBEI8AABgBgEAYAcBAGAIAQFgCQEAAAoAYAsBAWAMAQAADQEaAA4BAQAPAShgEAEBABEBOUASBD8rhR8AEwEBABQCA+hgFQEAYAYBAQAHAAAUAQJwGBxzZHJhbmdlbC5zYW1wbGVzb3VyY2UucnRsc2RycBkIMDAwMDAwMDEAGgCAG1KQAAEBAAIBVwADABAEAGAFAQFgBgEAAAcBAgAIAySfAGAJAQBgCgEAYAsBAGAMAQAgDQAQDgBgDwEAYBABAHARCTEyNy4wLjAuMRASAiK4EBMAcBwcc2RyYW5nZWwuc2FtcGxlc291cmNlLmhhY2tyZnAdEDkwOTg2NGM4Mzg4NTY5Y2YAHgCAH1KQAAEBAAEAYAMBARAEAQIABQECYAYBABAHAQwQCAMas/AQCQEWYAoBAWALAQAwDAOYloBgDgEAcA8JMTI3LjAuMC4xEBACIrgQEQBgEgEAIBMAAMgBAXDJFHNkcmFuZ2VsLmNoYW5uZWwuYmZtgcoBBJAAAQEAAQMMNQAAAgEHAAMBDwAEATIABQHEEAcE/wAA/4AIYZAAAQEAAQIIAAACAAADAQNABATCDAAAQAUEQkgAAGAGAQBgBwEAYAgBAGAJAQAACgBgCwEBYAwBAAANARYADgEBAA8BGmAQAQEAEQE+QBIEPyj1wwATAQEAFAIAyGAVAQBgCQEBYAoBAYALKZAAAQEAAQMMNQAQAgT/AAD/cAMSQnJvYWRjYXN0IEZNIERlbW9kAAcAcAwSQnJvYWRjYXN0IEZNIERlbW9kcA0VU3lzdGVtIGRlZmF1bHQgZGV2aWNlYA4BAHAPCTEyNy4wLjAuMRAQAhy7EBEAEBIAABMA

f4exb, 2020-07-18:
Looks OK but changed the title to something more explicit

f4exb, 2020-07-18:
The closed issue with the initial information: https://github.com/f4exb/sdrangel/issues/572

f4exb, 2020-07-18:
Works perfectly in my case in a Virtualbox VM with 20.04 and latest sdrangel .deb installed (similar as yours). There happens to be a station on the same frequency in my area:
[image]

What you may need to do is remove the current preset. I had trouble removing the existing channel that was there before trying to import your preset.

In order to do this with sdrangel not running:
 - edit `~/.config/f4exb/SDRangel.conf` file. 
 - remove the line starting with `current=`. 
 - save

Start sdrangel.

choupette-change, 2020-07-18:
According to the tests that I was able to carry out, the deletion of the line "current = ....." works a few times but it is necessary to delete this line with each new launch of sdrangel. I tried with an rtlsdr and a hackrf One, I have the same problem. This segfault problem mainly appears when loading and launching (testing several times) another .prex see the screenshot attached. I attach the contents of SDRangel.conf
[image]

choupette-change, 2020-07-18:
I attach the contents of SDRangel.conf: 
[General]
ambe="@ByteArray(AAAABHicY2BgYAAAAAQAAQ==)"
audio="@ByteArray(AAAAanicm8DAyNjAyMIABA1MMUCSEYi1GIIZKhmKGUoYUhlyGRQYUoB0GkMiQylDDlAMwi9jyGRIBtIMuxuAOoQYDBmMGMwZ9BgMoNhQnY8BGTACADIODXM=)"
current="@ByteArray(AAAEKnichVNNaxNRFL135qWTtGkz+ai0WtMgtSBImlRF6KqU4k4oVFy5mDGZ2uBk8sxMrF2Zgq7cVBAEN/ojXHSrIFjxB7jThb9B/EiJ570kOkrFN5y8m3fvue/ee2b2iVmyVfe23I4fScMKWqXAbXoVUyQOPtCe4EdE1AfoCDBIL34C82lP/RM/8HOSGhSQpA5FdB22Rzu0Ti2q0W2GOw98Aj6CuaTii7H4TUT7QA12jPMNeDiC4ixQCMYgrk0uYhv6xIe9G2PyAwT7I/Rw8AXGS+AxvLPfadiGoZIWkKCNpCHwx/XUx8K2owwVOQNPi5p4XJReB+Nf0YP8fA1401NeVvwTiN3W3EA3HCv42SAabs3W9nvFG1aq9566pptw9yEYsZHULpPFqhCv00SrCbEmiZwxJscCkkAKxHFmZwJIw57kIk2BneGiY2PP8sXVnDhfWYkozwYVeNmZZpWCycJo2JAz82G97Qa3PL8cuk3pe2Gr06555a2G7zUC2YnkrBK/e+rCvn6P0uWlyAujMliNO7aBe01AMMlEqrp8uVzBU7XHjDMHtkVybu747O3Ih0OeTlYGq4r3pTt/Q3eumiZbkJNAYjRLFipPmgvPdbvOODABoN3SJNlT5uLZF04GBzZqyMZqyKka8uQUkGYaed+i/MPCr3pq224QYL+51dx7x2IwdDN9CVPnMYw9Q4ILlOBD2xL9jXufu8mRMHZMmFdfB8Js/FeYxb+EWYEwRPePIIwJJQpamBTC0WB34tzvemxD3y/N3Fq75dZrbhiVrlwtrXvNVh0qyvQx53JyenM3jLxmafjZY7/bqHnOFIaUiQ3JVkPKkp1Tn/BPBCXMZQ==)"
hwDeviceUserArgs=@ByteArray(AAAAC3icm8DAyNjAyMIABAAJ5gEY)
limeRFEUSBCalib=@ByteArray(AAAAC3icm8DAyNjAyMIABAAJ5gEY)
preferences="@ByteArray(AAAAU3icm8DAyFjAJFOcUpSYl56ao1ecmFuQk1qcX1qUnKpXVJIDlChgZihgYWBgZXBgA1IMDA7sYIqBgyGBk5GhgIsHrjknP52BmwEAoHYTvw==)"

[preset-1]
data=@ByteArray(AAADy3ichVNNaxNRFL135qVJmq/JR6XREoLUgiBpUhShq1CKO6VQcCU442TaBieZYWbS2pXpwpUuKgiCG935F7pVEKz4A9y58UeIptTz3jR2EMUZbu7Nu+fcd97JyyEx+0y+VrjjhZHlbved+0HktnWROvpCB4KfEtEpgk4QGqmHX6B8OZHfxE98XKQ+DcmnEUV0F7VDe7ROHtn0kNGugqEjf0VelvhGAr8JtIuwUZ9zeAGwJ9OQnEUKwYhxAVnA9tWKi3o/yXwLsDuNCYTWUTyXgW79B50dQ5NDaxgQYGiISG6PI+NB2pOFRM6j49EArwXpPTD+hY7n688g5cNEdlnyLwC7o7hDdeCE4FcxGm3FVvVnyTtTqvJEbjNOWYf4vYi1TLwJi64Q7/NE3ZRY84nMGSYzjcggsiDOMps5RB51gRtUBLvEDdNALvP1bkVca69GVGWNarxizrEcwZSGNbgX8wthL7CG247bCq2B7zqhNwpsp4ULgoZfz7Tjp4MbML50T2mTssgQZKawLeRQGrMz+uJrJcicReQQENQskFHUl668MUtYMHAJy9nOys1WG2/HqGiXj4wqmdABWUwfoee49luPvWMNh8gPtgYHn1jEtuj5G/CFZ2BMiQTXKMXHRlqcbjz6Ns5MrTMS1r37Hlu38V/rlv6wbhXWET0+gXU6vKop67KA44Dj3NVzPYam9vf1ylrgWT3bCqPmrdvNdWfg9eCzn//Lul+Y29wPI2fQ7Dlb1siNkHf7tmMWYVIpYZIhTSqTUZF/sl/4G7ID)

[preset-2]
data="@ByteArray(AAADxXichVNNaxNRFL135qWTNF+Tj0qrEoLUgiAxKYrQVSjFnVCouBKccTK1wUkyzEysXZmCrtxUEAQ3+iNcdKsgWPEHuNOFv0H8SInnvWl0FKVvOPPuvHvPfeedvOwRs8/ka4bdDW8GkdfURWr/A+0KfkREE4AOAY3U4CcIn47ll/iB10nqUp98GlJE1xG7tE1rNCCH7jDSFeAT8BHMC7K+lqjfQLUHOIgTnG/AwykkZ5FCMOK6gGzUdtWKh3gnweQHKPamGGPhC4KXwGNkF77T0TE02bSKBgGahsAf29MEA9O2DGTlPDID6uGxIb0Dxv+q4/58DXgzllmW/BOo3VLcvjpwQvCzuBppxVbxe8k7UqrmsdxmlLL38EsRa2mV0lm0hXidI2qnxKpPZM0wWQaQBjIgzjJbWSCHOM81KoBd5JplYi7xxXZZnG+uRFRhjaq8bM2xbMFkwBrciPnTYSew+7ddrxHaPd9zw8EwcNwGLggS/kK6GY8WbsDo1A2lTcoiU5CVwraQQwZ6p/XF50qQNQtkAQiq58ks6EtnX1hFLJi4fqVMa/lyo4mnZZa1M/tmhSzogCymt9BzUP2lx9my+33MtzZ7u+9YxLbouUvwhWdgTJEEVynFB6YhJuv3Po/SU+vMhHWvvsbWrR9r3dJf1q3AOqL7h7BOh1dVZV0G5TjgKHvutx5TU/v7enk1GNgdxw6j+pWr9TW3N+jAZz/3j3U/P7exE0Zur95xN+2hF2G+23VcqwCTigmTTGlSicyy/JP9BGXTsXo=)"

f4exb, 2020-07-20:
There is no guarantee that changes made in plugins are upward compatible so some older presets may not work with a recent version of the plugin. Plugins are versioned and this can be a hint. This is probably what happens so you will have to create a new preset. Sometimes in the worst case scenario some offending parameter is saved in the current preset and starting the application subsequently will always fail. In this case you have to remove the current preset with the procedure described above.

choupette-change, 2020-07-20:
Thank you for the information but I uninstalled and reinstalled the latest version (v4.14.17) of Sdrangel by the .deb and I recreated the .prex in this version but I still have the same problem ???

f4exb, 2020-07-20:
We're chasing ghosts here. I do not understand at all what you are trying to do. This is not a general issue because I play with presets all the time and did not notice any specific issue. As said above it can happen sometimes that an old preset causes trouble but this is generally a rare case.

Firstly I think there is a confusion between a `.prex` file and a preset and the action of exporting and importing presets to and from `.prex` files and presets amd loading a preset into the current device set. It is not clear when the issue occurs:
  - at export time ? 
  - at import time ?
  - at preset load ?

I have no idea of what you have put in your presets moreover when I tried to paste the contents of your SDRangel.conf in mine it just cleared the preset information completely. So there is something weird with the preset data.

To get a different point of view can you use the REST API to retrieve the configuration information which is basically the contents of SDRangel.conf in readable form. If you start `sdrangel` with all the defaults it creates a web interface at http://127.0.0.1:8091/ so if you open it in a browser it will show a page with a "Interactive Swagger UI documentation". If you click on this it takes you to a second page where you can play with the API.

In this page:
  - click on "GET ​/sdrangel​/config" (it is the third link from the top)
  - then "Try it out' 
  - then "Execute" 
  - In the "Server response" par below under the "Response body" this is the reply from API in JSON format. 

Can you please copy + paste this reply preferably just before you encounter the issue telling me which preset you are trying to load? This will give me a more precise idea.

choupette-change, 2020-07-23:
I have uninstalled and reinstalled Ubuntu 20 04 and sdrangel_4.14.18-1_amd64.deb. Everything ok now, no cause identified ??? A question: is there a DAB radio plugin (such as Welle.io or Qt-Dab)? Thanks for the information given.

f4exb, 2020-08-05:
DAB issue already opened: https://github.com/f4exb/sdrangel/issues/169
This one can be closed.

closed 2020-08-05 by f4exb