f4exb/sdrangel #574: BP not working

countryboy057, 2020-07-17:
Hello 

I have installed the latest releases on both Windows and linux and neither are allowing BP to be decoded. Regular dmr works fine but BP is not. I have tried two different frequencies where I know the BP key.

How can I help troubleshoot this ?

f4exb, 2020-07-18:
BP is supposed to be implemented as per Motorola specs however I have no way to test this nor correct the codes if anything is wrong. The codes appear here (it isn dsdcc): https://github.com/f4exb/dsdcc/blob/master/dmr.cpp#L102 and is appliede here: https://github.com/f4exb/dsdcc/blob/master/dmr.cpp#L639

The full diff of the commit is here: https://github.com/f4exb/dsdcc/commit/9889e6a4f531650e55334e9e2c0312856b763a96 but also contains some cosmetic changes.

countryboy057, 2020-07-18:
I will get you a sample and upload it sometime today! Thank you!

countryboy057, 2020-07-18:
What is your email address? Id like to send the link to you that way.

f4exb, 2020-07-19:
please use f4exb06@free.fr

f4exb, 2020-07-21:
I have tried many variants in case of endianess mismatch or wrong lsb or msb first issue but none of the attempts leads to something intelligible. So the algorithm implemented is plain wrong and I have no way to correct this.

The solution is to remove this future in the next revision.

countryboy057, 2020-07-21:
There is another gentlemen that has implemented it . Maybe his code can
help ?

https://twitter.com/LouisErigHerve

His guthub

https://github.com/LouisErigHerve/dsd

I am a network admin and not very good with coding but am willing to help
if you ever need things tested and such

f4exb, 2020-07-21:
Thanks for the pointer that's interesting and seems quite different from my initial implementation.

countryboy057, 2020-07-21:
No problem. Thank your time in sdrangel and dsdcc. They are excellent
programs that I use dailey here.

f4exb, 2020-07-21:
Well checked the code of Louis Erig Herve but it is pretty empty. There are just comments on how things should work but no details on how those 882 bits (6*3*49) bits should be filled. Unfortunately this means I cannot carry on and will not support any kind of encryption in any way for any mode.

countryboy057, 2020-07-21:
I understand. I know there was some other documents out there also. I will
see if I can find them

countryboy057, 2020-07-21:
Here is a screenshot of somethings he was working on.

[image: image.png]

countryboy057, 2020-07-21:
[image: image.png]

countryboy057, 2020-07-21:
[image: image.png]

f4exb, 2020-07-21:
Maybe I overlooked the Twitter link it seems to contain much more information...

f4exb, 2020-07-21:
Well even like this it doesn't work.

countryboy057, 2020-07-21:
I was looking through your code. For the privacy keys.I am wondering if they need the 0x in front of them. The list I have for the keys just shows 4 hex characters.

f4exb, 2020-07-22:
`0x` prefix is the way in C/C++ (probably some other languages as well) to tell the compiler that what comes next are hexadecimal characters.

The issue is that the algorithm does not work for a reason I don't know and I don't want to dig more into this. I know people are excited about privacy codes well I am not.

countryboy057, 2020-07-22:
Could you share what you have redone and let me play around with it ? I
would like to learn more

TheDefender44, 2020-07-23:
> 
> 
> Thanks for the pointer that's interesting and seems quite different from my initial implementation.

I believe it is in part because his initial, non-public version includes auto-detection of the BP keys. That's why part of it is empty and/or quite different from your first implementation.

f4exb, 2020-07-23:
@countryboy057 the actual code is in dsdcc https://github.com/f4exb/dsdcc I have committed my latest changes to apply the XORing of bits only to the 49 data bits of the AMBE frame.

f4exb, 2020-08-01:
Now removed in v4.14.19

closed 2020-08-01 by f4exb