f4exb/sdrangel #578: ATV Modulator to match standards set in the Demodulator

f4exb, 2020-07-22:
Following the changes made recently to the ATV Demodulator the Modulator is out of sync regarding the proper settings of standard signal values set in the Demodulator. This issue is open to make the Modulator signal standard match and fix possible issues due to mismatch. It follows the discussion started at the end of #459

Vort, 2020-07-22:
Most important documentation is [Rec. ITU-R BT.1700](https://www.itu.int/dms_pubrec/itu-r/rec/bt/R-REC-BT.1700-0-200502-I!!ZPF-E.zip).
This article helps to understand how encoding should work: [Batsocks - Monochrome Composite Video](http://www.batsocks.co.uk/readme/video_timing.htm), but timings there are not accurate.

f4exb, 2020-07-22:
This is strange but the solution to the even/odd issue is deceivingly simple although the numbers do not seem to make sense (but it depends on how lines are actually counted):

<details><summary>Diff</summary>

```diff
diff --git a/plugins/channeltx/modatv/atvmodsource.cpp b/plugins/channeltx/modatv/atvmodsource.cpp
index 69ae3ff89..d0ee79553 100644
--- a/plugins/channeltx/modatv/atvmodsource.cpp
+++ b/plugins/channeltx/modatv/atvmodsource.cpp
@@ -590,7 +590,7 @@ void ATVModSource::applyStandard(const ATVModSettings& settings)
         m_nbImageLines     = m_nbLines - 15; // lines less the total number of sync lines
         m_nbImageLines2    = m_nbImageLines / 2;
         m_interlaced       = true;
-        m_nbSyncLinesHeadE = 5; // number of sync lines on the top of a frame even
+        m_nbSyncLinesHeadE = 4; // number of sync lines on the top of a frame even
         m_nbSyncLinesHeadO = 4; // number of sync lines on the top of a frame odd
         m_nbSyncLinesBottom = 3;
         m_nbLongSyncLines  = 2;
@@ -604,7 +604,7 @@ void ATVModSource::applyStandard(const ATVModSettings& settings)
         m_nbImageLines     = m_nbLines - 15;
         m_nbImageLines2    = m_nbImageLines / 2;
         m_interlaced       = true;
-        m_nbSyncLinesHeadE = 5;
+        m_nbSyncLinesHeadE = 4;
         m_nbSyncLinesHeadO = 4; // number of sync lines on the top of a frame odd
         m_nbSyncLinesBottom = 3;
         m_nbLongSyncLines  = 2;
@@ -619,7 +619,7 @@ void ATVModSource::applyStandard(const ATVModSettings& settings)
         m_nbImageLines     = m_nbLines - 15;
         m_nbImageLines2    = m_nbImageLines / 2;
         m_interlaced       = true;
-        m_nbSyncLinesHeadE = 5;
+        m_nbSyncLinesHeadE = 4;
         m_nbSyncLinesHeadO = 4; // number of sync lines on the top of a frame odd
         m_nbSyncLinesBottom = 3;
         m_nbLongSyncLines  = 2;
```

</details>

Now the zigzag displays correctly and some images are more crispy like this one:

<details><summary>Indian chart original</summary>

[image]

</detalis>

<details><summary>Indian chart received</summary>

[image]

</details>

f4exb, 2020-07-22:
This one corrects the timings

```diff
diff --git a/plugins/channeltx/modatv/atvmodsource.cpp b/plugins/channeltx/modatv/atvmodsource.cpp
index a8e004ac0..c956133cc 100644
--- a/plugins/channeltx/modatv/atvmodsource.cpp
+++ b/plugins/channeltx/modatv/atvmodsource.cpp
@@ -517,8 +517,8 @@ float ATVModSource::getRFBandwidthDivisor(ATVModSettings::ATVModulation modulati
 void ATVModSource::applyStandard(const ATVModSettings& settings)
 {
     m_pointsPerSync  = (uint32_t) ((4.7f / 64.0f) * m_pointsPerLine);
-    m_pointsPerBP    = (uint32_t) ((4.7f / 64.0f) * m_pointsPerLine);
-    m_pointsPerFP    = (uint32_t) ((2.6f / 64.0f) * m_pointsPerLine);
+    m_pointsPerBP    = (uint32_t) ((5.8f / 64.0f) * m_pointsPerLine);
+    m_pointsPerFP    = (uint32_t) ((1.5f / 64.0f) * m_pointsPerLine);
     m_pointsPerFSync = (uint32_t) ((2.3f / 64.0f) * m_pointsPerLine);
 
     m_pointsPerImgLine = m_pointsPerLine - m_pointsPerSync - m_pointsPerBP - m_pointsPerFP;
```

Vort, 2020-07-22:
You may noticed, that standard counts Sync and EQ pulses not in lines, but in half-lines.
So I think that this change should be merged (but I did not tested it yet - will do soon), but report should stay open until correct solution is implemented (which emits exactly correct amount of pulses for each mode).

f4exb, 2020-07-22:
The code is too old and clunky to be patched any further. To accomplish this it should be rewritten from the ground up.

Vort, 2020-07-22:
If you do not want to leave this report opened, new one can be created in future.
Right now I see only one small visible glitch: 
[image]
So this change is good for hack-fix.

Vort, 2020-07-22:
Oops. This "small glitch" produces crash in debug mode:
[image]

f4exb, 2020-07-22:
I think image is too short probably by one line since one line less counts as a sync line.

Guessing by previous values for all standards it seems the rule is the following:
`m_nbImageLines = m_nbLines - (m_nbSyncLinesHeadE + m_nbSyncLinesHeadO + m_nbSyncLinesBottom + m_nbHalfLongSync + m_nbWholeEqLine)`

Thus  `m_nbImageLines` should be `m_nbLines - 14` not `m_nbLines - 15`

This also explains the garbage at the bottom.

Vort, 2020-07-23:
> Thus m_nbImageLines should be m_nbLines - 14 not m_nbLines - 15

With this change, glitch and crash are gone.

f4exb, 2020-07-23:
Great! Thanks for confirming this. 

It is not strictly necessary to do a full rewrite and this concerns only the vertical sync lines. I don't think it is necessary to count the half lines either as long as you know what type of line should come next and this is an issue because the code as it is today is not well documented so it is hard to tell which parameter does what and one has to resort to voodoo magic to fix things. I will try and see if I can revise the vertical sync lines sequencing at minimal time cost and make code more explicit.

f4exb, 2020-08-01:
Released in v4.14.19 and 5.8.3

closed 2020-08-01 by f4exb

Vort, 2020-08-01:
Great work!
But looks like mistake slipped into VBI-819.pdf (which was detected by decoder):
[image]
Source: [VIIth Plenary Assembly of the C.C.I.R. (London, 1953): Volume I](http://search.itu.int/history/HistoryDigitalCollectionDocLibrary/4.273.43.en.100.pdf)

f4exb, 2020-08-01:
Looks like there is a swap of the fields. Then in the modulator what is counted as line 1 should become line 410 and line 409 should become line 1.

This is the current situation:
field1:
[image]

field2:
[image]

Vort, 2020-08-01:
> Looks like there is a swap of the fields.

At page 256: "Signal **at the end** of even fields", which means that odd field is presented.
So no problems here.

Most likely, it is enough to "replace" 408.5 with 409.5.

f4exb, 2020-08-01:
Indeed the end of one field is the start of the other... got it!

In atvmodsource.h:
For 625 lines
```c++
            else if (lineNumber == m_nbLines2)  // field transition 1 -> 2 (312)
            {
                return LineShortBroadPulses;
            }
```
For 819 lines:
```c++
            else if (lineNumber == m_nbLines2 - 1) // transition field 1 -> 2 (408)
            {
                return LineShortBroadPulses;
            }
```
I found the difference strange but did not challenge it. It looks like it follows the same logic as 625 lines.