f4exb/sdrangel #579: DATV Transport stream output of datv demodulator to UDP not displaying correctly (or not at all with VLC)

computer-wizard, 2020-07-24:
Release v4,14,18 on Ubuntu 20.04 (I had the same problem with older SDRangel releases)  
I can't receive UDP  transport stream output in VLC. With FFplay (FFmpeg) only the first 15 percent is displayed correctly  The rest of the picture is messed up (It displays perfectly in the video tab window of the datv demodulator in SDRangel!). 
Also no sound is produced by FFplay/FFmpeg.

This is some of the error output of FFplay/FFmpeg,  (DVB-S2 QPSK  2/3  Symbol rate 333k was used in this case)

mpegts @ 0x7f67f8000bc0] PES packet size mismatch
[aac @ 0x7f67f8567b80] Sample rate index in program config element does not match the sample rate index configured by the container.
[aac @ 0x7f67f8567b80] decode_pce: Input buffer exhausted before END element found
[aac @ 0x7f67f8567b80] channel element 2.13 is not allocated
[aac @ 0x7f67f8567b80] Sample rate index in program config element does not match the sample rate index configured by the container.
[aac @ 0x7f67f8567b80] decode_pce: Input buffer exhausted before END element found
[aac @ 0x7f67f8567b80] channel element 2.13 is not allocated
[h264 @ 0x7f67f8b18c80] left block unavailable for requested intra4x4 mode -1
[h264 @ 0x7f67f8b18c80] error while decoding MB 0 6, bytestream 18102
[h264 @ 0x7f67f8b18c80] concealing 1729 DC, 1729 AC, 1729 MV errors in I frame
[h264 @ 0x7f67f8dfa4c0] error while decoding MB 52 30, bytestream -6
[h264 @ 0x7f67f8dfa4c0] Cannot use next picture in error concealment
[h264 @ 0x7f67f8dfa4c0] concealing 237 DC, 237 AC, 237 MV errors in P frame
[aac @ 0x7f67f8567b80] Sample rate index in program config element does not match the sample rate index configured by the container.
[aac @ 0x7f67f8567b80] decode_pce: Input buffer exhausted before END element found
[aac @ 0x7f67f8567b80] channel element 2.13 is not allocated
[h264 @ 0x7f67f8d173c0] error while decoding MB 45 30, bytestream -6
[h264 @ 0x7f67f8d173c0] concealing 244 DC, 244 AC, 244 MV errors in P frame
[mpegts @ 0x7f67f8000bc0] PES packet size mismatchsq=    0B f=0/0   
[h264 @ 0x7f67f8b18c80] Reference 6 >= 2q=    7KB sq=    0B f=0/0   
[h264 @ 0x7f67f8b18c80] error while decoding MB 12 20, bytestream 201
[h264 @ 0x7f67f8b18c80] concealing 877 DC, 877 AC, 877 MV errors in P frame
[aac @ 0x7f67f8567b80] Sample rate index in program config element does not match the sample rate index configured by the container.
[aac @ 0x7f67f8567b80] decode_pce: Input buffer exhausted before END element found
[aac @ 0x7f67f8567b80] channel element 2.13 is not allocated
[h264 @ 0x7f67f8cfaac0] Reference 2 >= 2q=    9KB sq=    0B f=0/0   
[h264 @ 0x7f67f8cfaac0] error while decoding MB 3 24, bytestream 673
[h264 @ 0x7f67f8cfaac0] concealing 646 DC, 646 AC, 646 MV errors in P frame
[aac @ 0x7f67f8567b80] Sample rate index in program config element does not match the sample rate index configured by the container.
[aac @ 0x7f67f8567b80] decode_pce: Input buffer exhausted before END element found
[aac @ 0x7f67f8567b80] channel element 2.13 is not allocated
[h264 @ 0x7f67f8b53a80] concealing 82 DC, 82 AC, 82 MV errors in P frame
[h264 @ 0x7f67f8b18c80] mmco: unref short failure sq=    0B f=0/0   
[h264 @ 0x7f67f8cc1900] concealing 217 DC, 217 AC, 217 MV errors in P frame
[h264 @ 0x7f67f8d33cc0] Reference 2 >= 2q=    3KB sq=    0B f=0/0   
[h264 @ 0x7f67f8d33cc0] error while decoding MB 33 2, bytestream 1113
[h264 @ 0x7f67f8d33cc0] Cannot use next picture in error concealment
[h264 @ 0x7f67f8d33cc0] concealing 1936 DC, 1936 AC, 1936 MV errors in P frame
etc.

73,
Bob VE3TOK

f4exb, 2020-08-29:
There could be an issue with this piece of code: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demoddatv/datvudpstream.cpp#L47

Because when `m_tsBlockIndex` reaches `m_tsBlocksPerFrame` then it goes to the else block but the current block is dropped. The effect is to drop one block every 7 blocks.

f4exb, 2020-08-30:
Delivered in v4.15.4 and v5.9.4

closed 2020-08-30 by f4exb

computer-wizard, 2020-08-31:
Makes sense  TS works again in VLC and FFplay/FFmpeg, thanks for fixing it Edouard!

Bob