f4exb/sdrangel #580: Sdrplay support (question and feedback)

alphafox02, 2020-07-26:
Sorry if this isn’t the appropriate place, but just wanted to let you know that the Rsp1a with sdrplay api 3.07 works nicely within the latest sdrangel in 20.04. Have you built in native support for sdrplay or is it using soapy?

alphafox02, 2020-07-26:
Just a quick additional note. The RS1A is visible and selectable, works great. The RSDuo on the other hand is not shown in the sampling devices drop down menu. If there’s anything I can do to help trouble shoot please let me know. I’ve installed sdrangel from the .deb package. Currently on 4.14.16.

f4exb, 2020-07-27:
> the Rsp1a with sdrplay api 3.07 works nicely within the latest sdrangel in 20.04. Have you built in native support for sdrplay or is it using soapy?

You should be able to tell if it is using Soapy or not. If using Soapy then the device name in the mentions Soapy explicitly. I would be surprised if it was the native plugin being used because this one uses my version of `libmirisdr` and not the API provided by SDRplay. So unless there is a mysterious binary compatibility this is not supposed to work.

alphafox02, 2020-07-27:
So it’s the latter as I used your .deb and there’s no soapy mentioned in the name. I’ll admit, at first glance it works well with a quick check of an fm demo and listening to music. I thought it was a bit strange that the drop down block had me going in chunks of the spectrum and the gain controls did not look familiar at all. It’ll sorta work until it doesn’t, then I found that if I go open something like cubicsdr and get the radio working again and then come back to sdrangel it’ll again work. 

So anyways, sorry for the false alarm. If you’re in a pinch and it’s all you have, it sort of works, but like you said I think it’s just by some weird chance. 

I take it your .deb package doesn’t come with soapy module nor look for them in the standard path that 20.04 installs them from repositories? Be great if it did or could as cubicsdr sees not only the sdrplay equipment but also the usrp b205 mini I have.

f4exb, 2020-07-27:
I do not want to drag any soapy dependency. You have to install the soapy modules needed by your hardware.

alphafox02, 2020-07-27:
I have all soapy modules installed already as cubicsdr and SigDigger uses them. I’ve looked over your source install directions. If I use the .deb package is sdrangel looking somewhere specific for soapy modules? I was just trying to avoid having double installs of certain things and/or avoid building sdrangel from source. 

That’s great news on the b205. I also have a bladerf xa4 and that does seem to work okay. I’ll spend more time testing it.

alphafox02, 2020-07-27:
Looking at this

“The support libraries are installed in /opt/install/SoapySDR/lib/SoapySDR/modules0.7 (for the 0.7 version).”

Makes me think maybe if I replicate the directory structure and/or symlink what I already have installed then maybe sdrangel would use my currently installed modules.

f4exb, 2020-07-27:
Let's be clear. These are the options to build and/or run the software:
  - build from source using these instructions: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux Soon to be redirected to an Ansible playbook
  - build a Docker image using this ancillary project: https://github.com/f4exb/sdrangel-docker
  - install the .deb package on a Ubuntu 20.04 (no other version) found in the releases

There are no other options. I don't know and don't want to know about nvm, snappy, flatpack or anything else.

alphafox02, 2020-07-27:
I wasn’t suggesting yet another way to install. I was asking if the .deb package is looking for currently installed soapy modules. Not suggesting you bundle them in to the .deb just wondering if it’s possible for it to look for and interact with currently installed soapymodules. 

At any rate I think I can accomplish what I need by building from source and manually telling cmake where each and every already installed library is at.

closed 2020-07-27 by alphafox02