f4exb/sdrangel #581: Issue with Audio Out from SoundCard

merlinladruid, 2020-07-28:
Hi f4exb/Edouard,

I am new to SDRAngel and GIT hub. So apologies if i am posting this in the wrong place. Please feel free to move it in the right section. I am having issues with setting up the sound device(computer) using SDR Angel. I have made sure PulseAudio is installed and levels are high, but sttill i dont know if i am missing anything. See attached snips that show settings in SDRAngel and PulseAudio. What am i forgetting to set? The build version of SDR Angel is in the screenshot. Let me know if you need anything else like logs etc. Thanks in Advance - Merlin

[image]
[image]

f4exb, 2020-07-28:
You obviously have some issues with your audio setup that are not in control. The "L16/429..." value means that the sample rate returned was -1 (error). It is also unusual to see only the system default device in the list. You should have some messages in the log that tell that the audio device cannot be opened or something like that. So you need to fix your audio as a first step.

merlinladruid, 2020-07-28:
Thanks for your swift response f4exb. Appreciate it. To get the logs, i am assuming you are referring to the log collection procedure mentioned here using docker?

https://github.com/f4exb/sdrangel-docker

Let me know, if i am headed in the right direction. Thanks again!

merlinladruid, 2020-07-29:
Hi Again f4exb,

Here is a log from SDR angel:

2020-07-28 19:41:00.394 (I) SDRangel 5.1.0 Qt 5.9.5 64b x86_64 Ubuntu Core 18 DSP Rx:24b Tx:16b PID 2144
2020-07-28 19:41:05.231 (D) MainWindow::closeEvent
2020-07-28 19:41:05.231 (D) MainWindow::savePresetSettings: preset [default | no name]
2020-07-28 19:41:05.232 (D) DeviceUISet::saveRxChannelSettings: saving channel [sdrangel.channel.wfmdemod]
2020-07-28 19:41:05.232 (D) DeviceAPI::saveSamplingDeviceSettings: serializing source sdrangel.samplesource.limesdr[0]: LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=00090726074F2C3B
**2020-07-28 19:41:05.233 (D) AudioDeviceManager::serialize
2020-07-28 19:41:05.233 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  4294967295  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1**

Full log attached
[sdrangel.log](https://github.com/f4exb/sdrangel/files/4992006/sdrangel.log)

f4exb, 2020-07-31:
There is something wrong in your audio setup and I cannot help you with this.

However the failing audio device marked not usable somehow. Later this yields to an erroneous sample rate and can have adverse consequences. In this case it should result in a disabled audio.

This situation appears clearly in the plugin baseband constructor. Although failure is logged and the`start()`method returns `false` the audio device sample rate is still being used:

```
2020-07-28 19:41:11.099 (D) WFMDemodBaseband::WFMDemodBaseband
2020-07-28 19:41:11.099 (D) AudioDeviceManager::addAudioSink: -1: 0x55a69fac51b0
2020-07-28 19:41:11.099 (W) AudioOutput::start: using system default device 
2020-07-28 19:41:11.099 (W) AudioOutput::start: format -1 Hz 2xS16LE audio/pcm not supported. Using:  sampleRate: -1 channelCount: -1 sampleSize: -1 codec:  byteOrder: LE sampleType: 0
2020-07-28 19:41:11.099 (W) AudioOutput::start: Audio device '' failed
2020-07-28 19:41:11.099 (D) WFMDemodSink::applyAudioSampleRate: 4294967295
```

f4exb, 2020-07-31:
Actually it is already "marked" unusable as the `-1` sample rate is stored in the device information and significantly means it is unusable. This value is returned here:  https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodwfm/wfmdemodbaseband.cpp#L44

But the `applyAudioSampleRate` method takes an `unsigned int` in input so it cannot test for negativeness and take appropriate acttion: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodwfm/wfmdemodsink.cpp#L151

A similar situation probably exists in all plugins using audio.

f4exb, 2020-08-01:
closed in favor of #585

closed 2020-08-01 by f4exb

whyteks, 2021-10-02:
Hey @merlinladruid and Edouard
I'll take a wild guess that you are running the sdrangel "snap".
I installed same, on Ubuntu 18.04. and I see exactly what is described above.

Now, I know next to zero about so-called "snapcraft", but it looks like a kind of chroot jail to me, 
So, I simply ran ldd on the sdrangel binary to know what libs it wants and then did:
`LD_LIBRARY_PATH=/snap/sdrangel/4/lib/sdrangel/:/snap/sdrangel/4/lib:/snap/sdrangel/4/usr/lib/x86_64-linux-gnu/:/snap/sdrangel/4/usr/lib/x86_64-linux-gnu/pulseaudio/ /snap/sdrangel/4/bin/sdrangel
`

and immediately I see: 

`2021-10-02 12:46:48.946 (I) SDRangel 5.9.0 Qt 5.9.5 64b x86_64 Ubuntu 18.04.5 LTS DSP Rx:24b Tx:16b PID 4159
2021-10-02 12:46:48.946 (I) MainSettings::MainSettings: settings file: format: 0 location: /home/user/.config/f4exb/SDRangel.conf
2021-10-02 12:46:48.946 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2021-10-02 12:46:49.047 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2021-10-02 12:46:49.047 (D) AudioDeviceManager::AudioDeviceManager: input device #1: pulse
2021-10-02 12:46:49.047 (D) AudioDeviceManager::AudioDeviceManager: input device #2: sysdefault:CARD=PCH

[ etc etc etc ]
`

Supposedly sdrangel has all the permissions or "connections" to run correctly inside snap, 

`Interface     Plug                   Slot           Notes
alsa          sdrangel:alsa          :alsa          manual
home          sdrangel:home          :home          -
network       sdrangel:network       :network       -
network-bind  sdrangel:network-bind  :network-bind  -
opengl        sdrangel:opengl        :opengl        -
pulseaudio    sdrangel:pulseaudio    :pulseaudio    manual
raw-usb       sdrangel:raw-usb       :raw-usb       manual
unity7        sdrangel:unity7        :unity7        -
wayland       sdrangel:wayland       :wayland       -
x11           sdrangel:x11           :x11           -
`

but it cannot access audio devices inside the snap. Maybe something to do with the pulse config files inside /etc in the snap? I really don't know enough about it. 

Anyway, I don't mean to hijack a closed thread on this project to discuss snapcraft + pulse, but this note might help others who come across the issue, or maybe @f4exb, if you maintain the snap, this is something that could be fixed? I'd be happy to test anything on my system for you.

As always, thanks for all the code!

f4exb, 2021-10-02:
I do not support snaps in any way in fact I have removed snap from my system as it interferes with the sudo settings. I can only discourage the use of snaps for SDRangel. You should rather compile from source: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux