f4exb/sdrangel #582: No waterfall rendering for Mobility Radeon HD 2600 (+patch)

Vort, 2020-07-28:
I saw reports about non-working waterfall somewhere.
And now I was able to debug this problem on my notebook.
This bug, most likely, is happening because of bugs in video drivers.

Textures in OpenGL can be created in two different ways:
With [glTexStorage2D](https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/glTexStorage2D.xhtml) or with [glTexImage2D](https://www.khronos.org/registry/OpenGL-Refpages/gl4/html/glTexImage2D.xhtml).
When Qt sees OpenGL version >= 4.2 or GL_ARB(EXT)_texture_storage support, it uses glTexStorage2D.
In other cases glTexImage2D is used.

The problem is that some drivers claims GL_ARB_texture_storage support, but implements it wrong:
glTexSubImage2D works only with full texture height, which makes it useless for partial updates of m_glShaderWaterfall. In cases when height in GLShaderTextured::subTexture call is, for example, equals to 1, glGetError() returns 1282 or GL_INVALID_OPERATION.

So it is possible to check if immutable support is broken and use mutable storage in such case.
But [my patch](https://github.com/Vort/sdrangel/commit/5ebb21c20c7af4682305e68236518a7139e5be60) is simpler: it just uses mutable storage, which, theoretically, will be slower on modern hardware.
Now it is possible to test it with different hardware, check it for mistakes, upgrade to use fast processing on good hardware and slow on bad hardware or even close this report, but remember that hackfix exists. It is also possible to make binaries from time to time with this fix for people with problematic drivers or GPU.

f4exb, 2020-07-29:
I don't like the idea of downgrading the software to support faulty or obsolete hardware however if this situation can be detected at constructor time then we could know which option to use in the subsequent calls to critical methods and support both cases. I don't really get which operation returns GL_INVALID_OPERATION when glTexStorage2D is used so for now I cannot think of a simple way to probe this.

Vort, 2020-07-29:
> I don't really get which operation returns GL_INVALID_OPERATION when glTexStorage2D is used so for now I cannot think of a simple way to probe this.

This one:
https://github.com/f4exb/sdrangel/blob/96078bc9b864fdd27ac97ac17315451c15bbeb8a/sdrgui/gui/glshadertextured.cpp#L88

Test can be not so hard:
Create 1x2 texture with immutable storage and update first half of it (1x1) with glTexSubImage2D.
Or even create 1x1 and update 0x0 (need to test if it will generate error).

> however if this situation can be detected at constructor time then we could know which option to use in the subsequent calls to critical methods and support both cases

I think that such tests can have their own abstraction.
Like what is done in Chromium:
https://chromium.googlesource.com/chromium/src/+/master/gpu/config/gpu_driver_bug_list.json
That requires architecture thinking, which I am not good at.

Vort, 2020-07-29:
Test can look like this:
```c++
bool SomeClass::UseImmutableStorage()
{
    QOpenGLContext* ctx = QOpenGLContext::currentContext();
    QSurfaceFormat sf = ctx->format();

    if (sf.version() >= qMakePair(4, 2)
        || ctx->hasExtension(QByteArrayLiteral("GL_ARB_texture_storage"))
        || ctx->hasExtension(QByteArrayLiteral("GL_EXT_texture_storage")))
    {
        void (QOPENGLF_APIENTRYP glTexStorage2D)(
            GLenum target, GLsizei levels, GLenum internalFormat, GLsizei width, GLsizei height);
        glTexStorage2D = reinterpret_cast<void (QOPENGLF_APIENTRYP)(
            GLenum, GLsizei, GLenum, GLsizei, GLsizei)>(ctx->getProcAddress("glTexStorage2D"));
        int data = 0;
        GLuint textureId;
        glGenTextures(1, &textureId);
        glBindTexture(GL_TEXTURE_2D, textureId);
        glTexStorage2D(GL_TEXTURE_2D, 1, GL_RGBA8, 1, 1);
        glTexSubImage2D(GL_TEXTURE_2D, 0, 0, 0, 0, 0, GL_RGBA, GL_UNSIGNED_BYTE, &data);
        GLenum err = glGetError();
        glDeleteTextures(1, &textureId);
        return err == GL_NO_ERROR;
    }
    return false;
}
```

Vort, 2020-08-03:
Without the use of `QOpenGLTexture` code in 45c727d can look much cleaner.
But anyway it works with both Mobility Radeon HD 2600 and Intel HD Graphics 4600 now. Thanks.

f4exb, 2020-08-03:
Implemented in v4.14.20 and v5.8.5

closed 2020-08-03 by f4exb