f4exb/sdrangel #583: Connect to a SopayRemote

7Grok, 2020-07-30:
Hi

For the life of me I can't figure out how to make and use the eqivalent of a connection string. If my laptop is on the same network as the SopayRemote Sever Everything Just Work™, but I would like the ability to enter the remote address. Is there an equivalent of `soapy=0,driver=remote,remote-tcp://server:port`

As an FYI I compiled sdrangel on Mac OS by following the wiki. The only change I made was disabling devices that I don't have and it all works. My Bladerf via local usb and via SopayRemote works. Love the program and the docs any thoughts on setting up a pipeline to verify Mac OS compile?

--
Thanks
André

7Grok, 2020-07-31:
Progress .. at least with the custom device menu. I no receive an error that says cant open device, but not much in the debug logs.

closed 2021-03-17 by 7Grok

synthead, 2021-09-19:
Did you ever find a solution to this issue?

7Grok, 2021-09-20:
No. Sorry. I found an old issue that explains why if you feel like digging.