f4exb/sdrangel #589: plutosdr: v0.32 and sdrangel on windows/linux -- rx audio stutter; tx not tested.

roelandjansen, 2020-08-08:
https://github.com/analogdevicesinc/plutosdr-fw/issues/53

I have recently installed sdrangel 4.14.20 on windows and compiled from source on linux the same version.
I also updated the f/w to 0.32. Flashing takes pretty long compared to 0.31 (as in, maybe 4x longer?)

Anyways, 0.32 gives a stuttering sound ; reverting back to 0.31 gives a smooth sound.

Not sure if this is sdrangel related or v0.32 related or both.

roelandjansen, 2020-08-08:
https://github.com/analogdevicesinc/plutosdr-fw/issues/53

f4exb, 2020-08-16:
Just pulled out my Pluto with older firmware and I cannot make it work. It strangely displays the maximum possible frequency and cannot be moved from there. Not sure this is the same issue.

f4exb, 2020-08-16:
I confirm that after upgrade to FW 0.32 Pluto does not work properly with SDRangel even with latest libiio (v0.21).

f4exb, 2020-08-17:
It seems not to be in control of the code in SDRangel or something has changed that would require an update but only possile with guidance of the Pluto/libiio team. So for the time being stay with FW 0.31 on the Pluto.

f4exb, 2020-08-19:
Closed in favor of #604

closed 2020-08-19 by f4exb