f4exb/sdrangel #592: No save.

gaspar1987, 2020-08-09:
Hello sir, I have a problem with the new recording software. The problem is that it doesn't record and I'd like to know how you can show me how to fix it.
On the file and all that, that's it.
Thanks to see you soon.

f4exb, 2020-08-11:
Documentation is here https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/filesink/readme.md (as mentioned in the release notes). You can either use the record button (13) or activate the squelched recording (12). One thing I did not insist on is that the output file should be set via the file dialog (14) and btw the screenshot is incorrect it should be a `.sdriq` extension. If the field is empty then nething will happen.

gaspar1987, 2020-08-23:
Bonjour Monsieur j'ai testé votre méthode de votre de votre façon de le faire mais ça ne fonctionne pas il faut regarder sur le JavaScript du logiciel si sur mon ordinateur s'il est compatible ou pas compatible.

f4exb, 2020-08-24:
What is the connection with JavaScript?

closed 2020-09-08 by f4exb