f4exb/sdrangel #594: ATV Demodulator: arbitrary sample rate support

Vort, 2020-08-11:
Present state of ATV Demodulator allows enabling of arbitrary sample rate support with minimal changes to the code.
Here is the hack (which is _not_ intended for merging), which shows needed changes: https://github.com/Vort/sdrangel/commit/d089307ede64e40b8ef45c116a8c20a77258534a.
With this change, [`hacktv_dec2.sdriq`](https://github.com/f4exb/sdrangel/files/4878621/hacktv_dec2.zip) decodes properly.

Despite ease of enabling, much work is needed to do this properly.
Lots of existing code expects sample rate to be limited.
There are `ATVDemodSettings::getBaseValues` code, which calculate correct amounts of points pre line; `m_interpolator`, which converts samperates; lots of [documentation](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodatv/readme.md) about selecting proper sample rate.
All these things should be changed or at least reconsidered.

And since it is architectural changes, they are better to be done by @f4exb.

f4exb, 2020-08-11:
Yes it has quite a few side effects but much towards simplification. I can do that part.

Vort, 2020-08-11:
Important improvement is reduced CPU usage in case of "not round" (don't know how to call it) sample rate.

f4exb, 2020-08-11:
Yes my plan is to remove the rational down-sampler and its software FIR filter. With the bandwidths usually implied the hardware filtering is enough to get rid of unwanted out of band signals. Some devices like the LimeSDR also have a FIR filter implemented in hardware. I will keep the FFT filter as an option for corner cases but this is indeed an option.

Vort, 2020-08-11:
I completely agree with this (remove of decimator(?) and keeping FFT).
Also in some cases filtering can be done with Local channel sink + LocalInput.

Vort, 2020-08-11:
I have found a little bug in my hackfix:
If branch should be executed in both cases (with or without HSync):
https://github.com/f4exb/sdrangel/blob/aa8f9f6805e656c68f763863fb1b122b2119caec/plugins/channelrx/demodatv/atvdemodsink.h#L253
With disabled HSync `m_hSyncShift` just will be zero.

Vort, 2020-08-21:
Overall I like recent changes.
Filter adjustement now more smooth.

But I think that some corrections are needed for `applySettings` function:
<details>
<summary>m_settings changes</summary>

```patch
diff --git a/plugins/channelrx/demodatv/atvdemodsink.cpp b/plugins/channelrx/demodatv/atvdemodsink.cpp
index 6d031d327..0f51d1a92 100644
--- a/plugins/channelrx/demodatv/atvdemodsink.cpp
+++ b/plugins/channelrx/demodatv/atvdemodsink.cpp
@@ -515,21 +515,22 @@ void ATVDemodSink::applySettings(const ATVDemodSettings& settings, bool force)
         unsigned int samplesPerLineNom;
         ATVDemodSettings::getBaseValues(m_channelSampleRate, settings.m_nbLines * settings.m_fps, samplesPerLineNom);
         m_samplesPerLine = samplesPerLineNom;
-		m_samplesPerLineFrac = (float)m_channelSampleRate / (m_settings.m_nbLines * m_settings.m_fps) - m_samplesPerLine;
-		m_ampAverage.resize(m_samplesPerLine * m_settings.m_nbLines * 2); // AGC average in two full images
+		m_samplesPerLineFrac = (float)m_channelSampleRate / (settings.m_nbLines * settings.m_fps) - m_samplesPerLine;
+		m_ampAverage.resize(m_samplesPerLine * settings.m_nbLines * 2); // AGC average in two full images
 
         qDebug() << "ATVDemodSink::applySettings:"
                 << " m_channelSampleRate: " << m_channelSampleRate
                 << " m_samplesPerLine:" << m_samplesPerLine
                 << " m_samplesPerLineFrac:" << m_samplesPerLineFrac;
 
-        applyStandard(m_channelSampleRate, settings.m_atvStd, ATVDemodSettings::getNominalLineTime(settings.m_nbLines, settings.m_fps));
+        applyStandard(m_channelSampleRate, settings.m_atvStd,
+            ATVDemodSettings::getNominalLineTime(settings.m_nbLines, settings.m_fps));
 
         if (m_registeredTVScreen)
         {
             m_registeredTVScreen->resizeTVScreen(
                 m_samplesPerLine - m_numberSamplesPerLineSignals,
-                m_settings.m_nbLines - m_numberOfBlackLines
+                settings.m_nbLines - m_numberOfBlackLines
             );
 			m_tvScreenBuffer = m_registeredTVScreen->getBackBuffer();
         }
```
</details>

f4exb, 2020-08-21:
Thanks for spotting this. I had noticed the issue but wasn't able to spot it although it is pretty obvious when you see it. A second pair of eyes is always useful.

f4exb, 2020-08-25:
Implemented in v4.15.1

closed 2020-08-25 by f4exb